Frix-x/klippain-shaketune #5: Module 'matplotlib.colors' has no attribute 'TwoSlopeNorm'

pila81, 2023-11-04:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v1.1.1-0-g83f51775

### Describe the bug and expected behavior

After running the 'BELTS_SHAPER_CALIBRATION' no graph is generated and the following error is displayed:

Traceback (most recent call last):
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 231, in
main()
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 215, in main
fig, png_filename = get_belts_graph()
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 89, in get_belts_graph
fig = belts_calibration(lognames, KLIPPER_FOLDER)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 598, in belts_calibration
plot_difference_spectrogram(ax2, datas[0], datas[1], signal1, signal2, similarity_factor, max_freq)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 477, in plot_difference_spectrogram
norm = matplotlib.colors.TwoSlopeNorm(vcenter=np.min(combined_data), vmax=np.max(combined_data))
AttributeError: module 'matplotlib.colors' has no attribute 'TwoSlopeNorm'

Frix-x, 2023-11-06:
Hello, this error is due to your system using a very very old version of matplotlib: `TwoSlopeNorm` is a feature that was added almost 4 years ago now. Try to update the library using pip and it should be fixed :)

What OS are you running?

pila81, 2023-11-06:
The OS is Raspbian GNU/Linux 10 (buster) and according to Mainsails Update Manager everything was up-to-date.

Command 'pip3 show matplotlib' showed version 3.0.2.
After running 'sudo pip3 install --upgrade matplotlib' it's been upgraded to version 3.5.3.

Now everything works, thanks for the help!

closed 2023-11-06 by pila81

Frix-x, 2023-11-06:
Ok, yes, it seems that Debian Buster does indeed have an older version of matplotlib. Bookworm (the latest Debian) is up to date, so I didn't encounter this problem on my end.

In the meantime, since a lot of people are still using Buster on their printers, I pushed a fix in v1.1.2 to avoid using TwoSlopNorm and defining my own custom norm that replicates the same behavior. This should allow older versions of matplotlib to be used :)