Frix-x/klippain-shaketune #7: ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.

AlexanderSidorenko, 2023-11-07:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v1.1.1, 83f517758a7690692c1f35c21acf8c448c719771

### Describe the bug and expected behavior

I am consistently hitting an error deep in the Python guts when trying to run klippain-shaketune on my set up.
I repros 100%, on both X and Y axes. I am attaching one file to repro, but I have more if needed.

Here's the repro steps and full stack trace:
```
pi@fluiddpi:~/printer_data/config $ ./K-ShakeTune/scripts/graph_shaper.py K-ShakeTune_results/inputshaper/resonances_20231106_220647_X.csv -o test.png
/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py:54: DeprecationWarning: Use setlocale(), getencoding() and getlocale() instead
  locale.setlocale(locale.LC_TIME, locale.getdefaultlocale())
Fitted shaper 'zv' frequency = 54.4 Hz (vibrations = 10.8%, smoothing ~= 0.058)
To avoid too much smoothing with 'zv', suggested max_accel <= 11500 mm/sec^2
Fitted shaper 'mzv' frequency = 36.0 Hz (vibrations = 2.3%, smoothing ~= 0.157)
To avoid too much smoothing with 'mzv', suggested max_accel <= 3800 mm/sec^2
Fitted shaper 'ei' frequency = 50.4 Hz (vibrations = 0.0%, smoothing ~= 0.127)
To avoid too much smoothing with 'ei', suggested max_accel <= 4700 mm/sec^2
Fitted shaper '2hump_ei' frequency = 54.4 Hz (vibrations = 0.0%, smoothing ~= 0.182)
To avoid too much smoothing with '2hump_ei', suggested max_accel <= 3300 mm/sec^2
Fitted shaper '3hump_ei' frequency = 93.8 Hz (vibrations = 0.0%, smoothing ~= 0.093)
To avoid too much smoothing with '3hump_ei', suggested max_accel <= 6400 mm/sec^2
Recommended shaper is 3hump_ei @ 93.8 Hz
Axis has a main resonant frequency at 55.8Hz with an estimated damping ratio of 0.098
Peaks detected on the graph: 2 @ 55.8, 130.8 Hz (2 above effect threshold)
Traceback (most recent call last):
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 395, in <module>
    main()
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 390, in main
    fig = shaper_calibration(args, options.klipperdir, options.max_smoothing, options.max_freq)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 356, in shaper_calibration
    plot_spectrogram(ax2, datas[0], peaks, max_freq)
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 276, in plot_spectrogram
    pdata, bins, t = compute_spectrogram(data)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 124, in compute_spectrogram
    pdata, bins, t = _specgram(d['x'])
                     ^^^^^^^^^^^^^^^^^
  File "/home/pi/printer_data/config/./K-ShakeTune/scripts/graph_shaper.py", line 119, in _specgram
    return matplotlib.mlab.specgram(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 780, in specgram
    spec, freqs, t = _spectral_helper(x=x, y=None, NFFT=NFFT, Fs=Fs,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 382, in _spectral_helper
    result = _stride_windows(x, NFFT, noverlap)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 258, in _stride_windows
    return np.lib.stride_tricks.sliding_window_view(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<__array_function__ internals>", line 200, in sliding_window_view
  File "/usr/lib/python3/dist-packages/numpy/lib/stride_tricks.py", line 336, in sliding_window_view
    return as_strided(x, strides=out_strides, shape=out_shape,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/numpy/lib/stride_tricks.py", line 105, in as_strided
    array = np.asarray(DummyArray(interface, base=x))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
pi@fluiddpi:~/printer_data/config $ cd K-ShakeTune
pi@fluiddpi:~/printer_data/config/K-ShakeTune $ git status
On branch main
Your branch is up to date with 'origin/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	scripts/__pycache__/

nothing added to commit but untracked files present (use "git add" to track)
pi@fluiddpi:~/printer_data/config/K-ShakeTune $ git show HEAD
commit 83f517758a7690692c1f35c21acf8c448c719771 (HEAD -> main, tag: v1.1.1, origin/main, origin/HEAD)
Merge: 5366ad0 c156459
Author: Félix Boisselier <accounts@fboisselier.fr>
Date:   Wed Nov 1 20:09:50 2023 +0100

    Merge pull request #4 from Frix-x/develop
    
    v1.1.1

pi@fluiddpi:~/printer_data/config/K-ShakeTune $ 
```

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/13276456/klippy.log)
[resonances_20231106_220647_X.zip](https://github.com/Frix-x/klippain-shaketune/files/13276477/resonances_20231106_220647_X.zip)

Frix-x, 2023-11-08:
Hello,

Thanks for raising an issue about this. It's indeed an identified bug. For now your are the third one to have it and I'm still unsure about what is the origin.
What I know is that it's not coming directly from my scripts but from the plotting library that fail to allocate RAM to create the spectrogram. However, by doing some very basic profiling, the array is around 15-20MB and this should work on about every single board computer. Beside that, I know that some users are running my scripts fine in a Pi Zero so it's definitely something to be investigated to understand why it fail sometimes...

AlexanderSidorenko, 2023-11-09:
Thanks for looking into this!
I just checked, and my board has 4 GB of RAM. Most of it is free.
I also bumped swap from 100MB to 4GB as an experiment, but I get the same result.
I will look into this if/when I have spare cycles, since this got me really curious, and I do want to have Klippain working on my set up.

One more interesting/weird data point: Klippain worked on the same HW, just with different OS installation. I used fluiddpi originally, but upgraded to the latest Raspberry Pi OS + KIAUH recently, and it now fails all the time.

recrudesce, 2023-11-10:
Likely due to integer overflow on 32 bit machines.

@AlexanderSidorenko are you running 32bit OS perchance ?

AlexanderSidorenko, 2023-11-11:
Yes, actually! It is in fact 32 bit OS, and that is the difference between my original setup where Klippain worked, and the current one where things are broken.

Frix-x, 2023-11-12:
Hi, I've taken some time to do some tests on my machines and also on some machines of contributors who helped me.

So here are the items I got so far:
1. At the moment I got only 3 reports of this problem and I estimate around 100-300 people using it right now and this number is increasing everyday (I got between 20 to 50 unique cloners everyday). So in this number of people there are most likely both 32bit and 64bit systems.
2. On my side, I took some time to test the script on 64bit environments (a Pi3, a Pi4, a Fedora machine and in WSL) as well as 32bit environments (a PiZero2W, a Pi3, a Pi4 and a virtual machine running Debian 32bits) and it was ok on all of them. I also tried using the problematic CSV sent by the users experiencing the problem, but could not reproduce it on my end. So I think I can confirm that it's *probably* not a 32bit overflow problem.
3. Among the contributors who helped me debug the script, one experienced the problem (he is part of the 3 reports), but all other people have no problems.
4. I did some memory profiling and also tried to see if I could reduce the memory usage of the spectrogram generation by reducing the FFT window and overlap. This made a very small improvement, but apparently not enough to fix the problem (but not sure if the problem is really coming from there).
5. I also tried forcing the use of float32 instead of float64, but this made almost no change (it was actually a little worse) to the memory usage, so I think it's due to how the numbers are handled in the matplotlib library instead of my script.
6. The profiling gave me some interesting results, confirming that the memory consumption is quite low in relation to the 1-4 GB of RAM of our systems, since the full script run is about 100 MB of RAM and the problematic function call is about 30-50 MB of RAM.

Frix-x, 2023-11-12:
For reference, here is one of the worst report that I got regarding the spectrogram generation:

```
Line #    Mem usage    Increment  Occurrences   Line Contents
=============================================================
   112   77.277 MiB   77.277 MiB           1   @profile
   113                                         def compute_spectrogram(data):
   114   77.277 MiB    0.000 MiB           1       N = data.shape[0]
   115   77.277 MiB    0.000 MiB           1       Fs = N / (data[-1,0] - data[0,0])
   116                                             # Round up to a power of 2 for faster FFT
   117   77.277 MiB    0.000 MiB           1       M = 1 << int(.5 * Fs - 1).bit_length()
   118   77.277 MiB    0.000 MiB           1       window = np.kaiser(M, 6.)
   119   84.523 MiB    0.000 MiB           4       def _specgram(x):
   120  109.379 MiB   56.957 MiB           6           return matplotlib.mlab.specgram(
   121   84.523 MiB  -24.855 MiB           3                   x, Fs=Fs, NFFT=M, noverlap=M//2, window=window,
   122   84.523 MiB    0.000 MiB           3                   mode='psd', detrend='mean', scale_by_freq=False)
   123                                         
   124   77.277 MiB    0.000 MiB           1       d = {'x': data[:,1], 'y': data[:,2], 'z': data[:,3]}
   125   84.523 MiB    0.000 MiB           1       pdata, bins, t = _specgram(d['x'])
   126   84.523 MiB    0.000 MiB           3       for ax in 'yz':
   127   84.523 MiB  -49.711 MiB           2           pdata += _specgram(d[ax])[0]
   128   84.523 MiB    0.000 MiB           1       return pdata, bins, t
```

Frix-x, 2023-11-12:
For users having the problem here, can you try running this in your SSH console: `python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'` ?

ukutghost, 2023-11-12:
Morning, 

Running that gives me on my V2 where I have the issue and VT where I don't gives the same result:

``` 
➜  ~ python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'
7fffffff False
```

Both are running 32bit raspbian

ukutghost, 2023-11-12:
Version checks on bits

Python
V2 (bug present): 3.9.2
VT (working fine): 3.9.2

V2: 1.25.2
VT: 1.19.5

Matplotlib
V2: 3.7.2
VT: 3.3.4

AlexanderSidorenko, 2023-11-12:
> For users having the problem here, can you try running this line in your SSH console: `python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'` ?

```
pi@fluiddpi:~ $ python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'
7fffffff False
```

stevereno30, 2023-11-24:
I also have this issue on a Pi4b running Raspbian lite. It is a recent installation which I ran through using directions I wrote for myself the last time I clean installed Klipper on my old Pi3b in October. This issue didn't exist on my 3b. After running the command above, I get this:
python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'
7fffffff False

stevereno30, 2023-11-26:
Installing Debian Buster (legacy Raspberry Pi OS) fixed this issue for me. The current version of RPiOS is based on Debian Bookworm, and it is apparently incompatible with some part of Shake & Tune.

ukutghost, 2023-11-26:
I'm running Debian Bullseye on both printers with Shake&Tune installed, only one of them has the issue

Frix-x, 2023-11-28:
So based on all of the discussions, feedbacks and testings, it look like it's not really a 32bit vs 64bit issue since Shake&Tune was confirmed to work on both of this kind of systems. However, the issue seems to arise only on 32 bits system AFAIK.

Still looking to find the origin. If I can't solve it I will probably switch to another plotting librairy like plotly or something like that in the future... Let's see!

First thing to try for now is to make the whole thing run in a Python env and see if I can fix the lib versions and the bug that way

supahobnobba, 2023-11-28:
> So based on all of the discussions, feedbacks and testings, it look like it's not really a 32bit vs 64bit issue since Shake&Tune was confirmed to work on both of this kind of systems. However, the [...]

Also, seeing the same error. 
[image]

Thanks for all your efforts, Frix! Looking forward to trying this out for the first time.

amdavidson, 2023-12-07:
Chiming in to say that I have this issue also... 

Uname:
```bash
$ uname -a
Linux bluey 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

Mainsail host data:
```
Host(aarch64, 32bit)
Version: v0.12.0-12-g38221df8-dirty
OS: Raspbian GNU/Linux 12 (bookworm)
Load: 1.49, Mem: 289.9 MB / 908.3 MB,  Temp: 37°C
wlan0 (192.168.6.56) : Bandwidth: 5.7 kB/s , Received: 220.4 MB , Transmitted: 4.0 GB
```

The test from the comments:
```
$ python3 -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'
7fffffff False
```

Traceback:
```
Traceback (most recent call last):
  File "/home/amd/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 246, in <module>

    main()
  File "/home/amd/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 230, in main

    fig, png_filename = get_belts_graph()
                        ^^^^^^^^^^^^^^^^^
  File "/home/amd/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 98, in get_belts_graph

    fig = belts_calibration(lognames, KLIPPER_FOLDER)

          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/amd/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 603, in belts_calibration

    plot_difference_spectrogram(ax2, datas[0], datas[1], signal1, signal2, similarity_factor, max_freq)

  File "/home/amd/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 467, in plot_difference_spectrogram

    combined_data, bins, t = combined_spectrogram(data1, data2)

                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/amd/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 298, in combined_spectrogram

    pdata1, bins1, t1 = compute_spectrogram(data1)

                        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/amd/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 185, in compute_spectrogram

    pdata, bins, t = _specgram(d['x'])

                     ^^^^^^^^^^^^^^^^^
  File "/home/amd/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 180, in _specgram
    return matplotlib.mlab.specgram(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 780, in specgram

    spec, freqs, t = _spectral_helper(x=x, y=None, NFFT=NFFT, Fs=Fs,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 382, in _spectral_helper
    result = _stride_windows(x, NFFT, noverlap)

             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/matplotlib/mlab.py", line 258, in _stride_windows

    return np.lib.stride_tricks.sliding_window_view(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<__array_function__ internals>", line 200, in sliding_window_view
  File "/usr/lib/python3/dist-packages/numpy/lib/stride_tricks.py", line 336, in sliding_window_view

    return as_strided(x, strides=out_strides, shape=out_shape,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3/dist-packages/numpy/lib/stride_tricks.py", line 105, in as_strided
    array = np.asarray(DummyArray(interface, base=x))

            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.

Stats 10705.7: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000005 mcu_task_stddev=0.000004 bytes_write=1958767 bytes_read=302061 bytes_retransmit=9 bytes_invalid=0 send_seq=34641 receive_seq=34641 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=180000097 rpi: mcu_awake=0.184 mcu_task_avg=0.000030 mcu_task_stddev=0.000011 bytes_write=21129 bytes_read=5055272 bytes_retransmit=0 bytes_invalid=0 send_seq=3058 receive_seq=3058 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999566 adj=49999641  heater_bed: target=0 temp=22.5 pwm=0.000 chamber: temp=21.6 sysload=1.20 cputime=45.382 memavail=576404 print_time=361.606 buffer_time=0.000 print_stall=0 extruder: target=0 temp=24.5 pwm=0.000
Command {plot_graph} finished
```

Ekodas, 2023-12-07:
Also having this issue with fresh installs across the board.  Posted an issue in the Klippain github at https://github.com/Frix-x/klippain/issues/379 since I've installed the complete klippain, not just the Shake n Tune.

Ekodas, 2023-12-07:
Can confirm, changing to 64-bit Raspberry Pi OS Lite fixed the issue.  Just finished my installs and testing and I'm now getting the graphs.

Frix-x, 2023-12-10:
Hey,

I've worked on this subject and may have found a way to fix it. So if some of you could test it, I would be glad :)
This fix need a new library that is not included in default Klipper measuring resonances install guide so the whole Shake&Tune package was converted to work in a venv (but it's also cleaner at the end as I can manage the installation of everything more easily)

To proceed with the test:
  1. You will need to get the "venv" branch. In SSH type the following:
  ```bash
  cd ~/klippain_shaketune
  git checkout venv
  ```
  2. Then you will need to start the install script again manually to create the venv
  ```bash
  ./install.sh
  ```
  3. Run the problematic macro and confirm that it's now working for you :)

FYI, these manual steps are only necessary now since I want you to test it before creating a release tag, but when it will be pushed officially to the main branch and released, updating Shake&Tune from Mainsail/Fluidd will be enough for everyone and everything will be automated :)

sitofabi, 2023-12-10:
This fix works for me on bookworm and a PI5. 
Had this issue also on a 32bit bookworm installation with my Pi5. Checked out the venv and setup everything manually due to the fact that my whole directory structure is different. 
I had to apt-install libopenblas-dev and libatlas-base-dev but then everything worked without any error 👍

Frix-x, 2023-12-10:
Wouhouu that's great news!

I'll wait for the confirmation by other people and will eventually release it later today for everyone :)

ukutghost, 2023-12-10:
Can confirm, that fix has worked for me as well

Frix-x, 2023-12-10:
Ok that's perfect :)
I'll push the release ASAP!

commit 70500182 (2023-12-10): fixed #7 and code optimizations
closed 2023-12-11 by Frix-x