Frix-x/klippain-shaketune #8: Ubuntu server  -- PermissionError: [Errno 13] Permission denied: '/proc/1630/fd/0'

DocHopper, 2023-11-19:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v1.1.2-0-g6e884528

### Describe the bug and expected behavior

When I run any of the calibrations, at the time of plot creation, I get the following messages. They vary based on the type of calibration, but the common thread is the permission denied error. 

```
Command {plot_graph} finished
File "/home/doc/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 231, in
main()
File "/home/doc/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 217, in main
fig, png_filename = get_shaper_graph()
File "/home/doc/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 107, in get_shaper_graph
while is_file_open(filename):
File "/home/doc/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 55, in is_file_open
if os.path.samefile(fd, filepath):
File "/usr/lib/python3.10/genericpath.py", line 100, in samefile
s1 = os.stat(f1)
PermissionError: [Errno 13] Permission denied: '/proc/1630/fd/0'
Traceback (most recent call last):
Running Command {plot_graph}...:
X axis shaper graphs generation...
Writing raw accelerometer data to /tmp/raw_data_x_x.csv file
``` 

### Additional information and klippy.log

[klippy (11).log](https://github.com/Frix-x/klippain-shaketune/files/13403268/klippy.11.log)

Frix-x, 2023-11-19:
A potential fix is available in this commit: https://github.com/Frix-x/klippain-shaketune/commit/a4c2ead732ba7ba5daca371b0856f23e6cb0a834

Let me know if it's working correctly. I'll do some regression test on my side also

DocHopper, 2023-11-19:
Thanks Frix!  This seemed to have done the trick!  Nice work!

Frix-x, 2023-11-24:
Hi,

I'm writing to let you know about some changes I made to fix issue #13. I went back to the old way of checking when Klipper is done writing the CSV file because the new method implemented to fix your error didn't work well and get the original problem back: sometimes Klipper wasn't finished writing the file when my script started using it, and this made the file unusable (as experienced by the user in #13).

But to not reintroduce your error at the same time, I also added something new to handle permissions errors. This means the script shouldn't crash anymore if it tries to access a filedescriptor your user can't read. Could you try the latest commit and let me know if everything still works fine you?

Thanks for your help with this!

PR #9 (2023-11-20): using fcntl to check if a file is still open by klipper
This is to fix #8 
ie. switched to `fcntl` to try to lock a file instead of just crawling `/proc/.../fd` that can lead to permissions errors on some systems.
closed 2023-11-26 by Frix-x