Frix-x/klippain-shaketune #10: Very different results compared to klipper built in tools?

celtare21, 2023-11-20:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v1.1.2

### Describe the bug and expected behavior

I get extremely different recommendations from running "AXES_SHAPER_CALIBRATION" compared to "SHAPER_CALIBRATE" but only on the X axis.
on the X axis with AXES_SHAPER_CALIBRATION the recommendations are:

```
"// Writing raw accelerometer data to /tmp/raw_data_x_x.csv file
16:54:06 
echo: X axis shaper graphs generation...
16:54:06 
// Running Command {plot_graph}...:
16:54:19 
// Fitted shaper 'zv' frequency = 53.8 Hz (vibrations = 17.0%, smoothing ~= 0.060)
// To avoid too much smoothing with 'zv', suggested max_accel <= 11300 mm/sec^2
// Fitted shaper 'mzv' frequency = 34.0 Hz (vibrations = 2.6%, smoothing ~= 0.176)
// To avoid too much smoothing with 'mzv', suggested max_accel <= 3400 mm/sec^2
// Fitted shaper 'ei' frequency = 49.6 Hz (vibrations = 2.5%, smoothing ~= 0.131)
// To avoid too much smoothing with 'ei', suggested max_accel <= 4600 mm/sec^2
// Fitted shaper '2hump_ei' frequency = 41.4 Hz (vibrations = 0.0%, smoothing ~= 0.315)
// To avoid too much smoothing with '2hump_ei', suggested max_accel <= 1800 mm/sec^2
// Fitted shaper '3hump_ei' frequency = 57.4 Hz (vibrations = 0.0%, smoothing ~= 0.249)
// To avoid too much smoothing with '3hump_ei', suggested max_accel <= 2300 mm/sec^2
// Recommended shaper is ei @ 49.6 Hz
// Axis has a main resonant frequency at 55.7Hz with an estimated damping ratio of 0.049
// Peaks detected on the graph: 3 @ 43.6, 55.7, 85.4 Hz (2 above effect threshold)
// Graphs created. You will find the results in /home/pi/printer_data/config/K-ShakeTune_results"

```
while on INPUT_SHAPER I get:
```
// Re-enabled [input_shaper]
16:50:55 
// Calculating the best input shaper parameters for x axis
16:50:56 
// Fitted shaper 'zv' frequency = 81.6 Hz (vibrations = 1.6%, smoothing ~= 0.030)
16:50:56 
// To avoid too much smoothing with 'zv', suggested max_accel <= 25900 mm/sec^2
16:50:58 
// Fitted shaper 'mzv' frequency = 77.6 Hz (vibrations = 0.0%, smoothing ~= 0.036)
16:50:58 
// To avoid too much smoothing with 'mzv', suggested max_accel <= 17700 mm/sec^2
16:50:59 
// Fitted shaper 'ei' frequency = 93.6 Hz (vibrations = 0.0%, smoothing ~= 0.038)
16:50:59 
// To avoid too much smoothing with 'ei', suggested max_accel <= 16300 mm/sec^2
16:51:00 
// Fitted shaper '2hump_ei' frequency = 118.8 Hz (vibrations = 0.0%, smoothing ~= 0.041)
16:51:00 
// To avoid too much smoothing with '2hump_ei', suggested max_accel <= 15700 mm/sec^2
16:51:02 
// Fitted shaper '3hump_ei' frequency = 145.8 Hz (vibrations = 0.0%, smoothing ~= 0.042)
16:51:02 
// To avoid too much smoothing with '3hump_ei', suggested max_accel <= 15600 mm/sec^2
16:51:02 
// Recommended shaper_type_x = mzv, shaper_freq_x = 77.6 Hz
```

but they are 1:1 on the Y axis.
The graphs from BELTS_SHAPER_CALIBRATION compared to
TEST_RESONANCES AXIS=1,1 OUTPUT=raw_data
TEST_RESONANCES AXIS=1,-1 OUTPUT=raw_data

are also quite different:
[image]
[image]

I don't know if it matters but by accelerometer doesn't match X,Y,Z, it's actually mounted as X,Z,Y but as far as I know this doesn't matter on Klipper.
Let me know if there's anything else useful I can attach

Frix-x, 2023-11-21:
Hello,

I'm puzzled as to why you're encountering this issue. My scripts essentially utilize Klipper's original shaper calibration tools under the hood, only adding some enhancements and cosmetics to the graphical output. 

The key distinction from Klipper's standard `SHAPER_CALIBRATE` lies in the ability to set a maximum smoothing parameter in your configuration. This can significantly influence the shaper choice that Klipper would ordinarily make using the more standard `TEST_RESONANCES` method. I suspect the discrepancy in smoothing values between the `SHAPER_CALIBRATE` results and those generated by my script – which again uses the same Klipper function for calculation – might be due to this smoothing parameter you've set.

Beside that, my script offers all potential options, including Klipper's choice as 'performance shaper' selection. Additionally, it presents a 'low vibration' alternative that often yields cleaner print results, should it differ from Klipper's choice.

To further investigate, could you please share your klippy.log file? This would help in confirming if the added parameter is causing these differences.

celtare21, 2023-11-21:
Hello!

That's why I remained puzzled too, I checked the scripts and as you say, they use klipper commands.
Also because for some reason, they output on Y in exactly the same in both cases, only X is different.

I have attached below 2 zips, one with the SHAPER_CALIBRATE data, including a fresh klippy.log (restart, home, run the calibration) and also the CSV files and one with the output from AXES_SHAPER_CALIBRATION.
[shaper_calibrate.zip](https://github.com/Frix-x/klippain-shaketune/files/13425242/shaper_calibrate.zip)
[axes_shaper_calibration.zip](https://github.com/Frix-x/klippain-shaketune/files/13425313/axes_shaper_calibration.zip)

Frix-x, 2023-11-24:
I've got someone else experiencing the same issue on the Voron Discord. So at least it's not something happening only at you.
I'll work on this this weekend or tonight.

celtare21, 2023-11-24:
Glad I'm not the only one 😅.
Awesome, hope you find something. If you need testing let me know, love those graphs

Frix-x, 2023-11-27:
Ok I may have found the problem and pushed a potential fix. Can you update to the latest commit from the main branch?
(use moonraker update manager `channel: dev` to be able to do this)

Frix-x, 2023-11-27:
> @Frix-x sure thing. Would this also affect the X Y AXES_SHAPER_CALIBRATION issue?

Yes, it's due to corrupted file or incomplete writes, because sometimes the OS didn't have time to write the whole CSV file on slow file systems. I already fixed the problem long time ago for the Klipper CSV generation part, but my script also moves the CSV file to another location. With Shake&Tune I now use a Python shutil.move command to do this, and the behavior is apparently a bit different than the OS `mv` because it's not synchronous. So I added another check and sync for the file system to avoid this again at this step. 

Lets see if this improves the behavior...

celtare21, 2023-11-27:
@Frix-x I can confirm that both AXES_SHAPER_CALIBRATION and BELTS_SHAPER_CALIBRATION work properly now, 1:1 with the normal klipper values.
Thanks a lot for the fix and great job!
I'll close this issue now as it's not longer an issue

closed 2023-11-27 by celtare21