Frix-x/klippain-shaketune #11: Accelerator name is hardcoded in get_vibtations_graph

JurajMik, 2023-11-21:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v 1.1.2 6e88452

### Describe the bug and expected behavior

get_vibrations_graph in is_workflow.py uses hardcoded accelerometer:
globbed_files = glob.glob('/tmp/adxl345-*.csv')

I do not have adxl345, instead USB lis2dw (biqu). Passing chip name as "hotend" it looks into lis2dw.cfg (I have two probes, one on hotend, one on bed) :

lis2dw.cfg
-----------
# This file contains common pin mappings for the bigtreetech lis2dw v1.0
# To use this config, the firmware should be compiled for the
# RP2040 with "USB"
# The micro-controller will be used to control the components on the nozzle.

# See docs/Config_Reference.md for a description of parameters.

[include ./SHAPER/shaper.cfg]

[mcu lis2dwHotend]
serial: /dev/serial/by-path/platform-5200000.usb-usb-0:1.3:1.0 

[mcu lis2dwBed]
serial: /dev/serial/by-path/platform-5200000.usb-usb-0:1.4:1.0

[lis2dw hotend]
cs_pin: lis2dwHotend:gpio9
#spi_bus: spi1a
spi_software_sclk_pin: lis2dwHotend:gpio10
spi_software_mosi_pin: lis2dwHotend:gpio11
spi_software_miso_pin: lis2dwHotend:gpio8
axes_map: -x,y,z

[lis2dw bed]
cs_pin: lis2dwBed:gpio9
#spi_bus: spi1a
spi_software_sclk_pin: lis2dwBed:gpio10
spi_software_mosi_pin: lis2dwBed:gpio11
spi_software_miso_pin: lis2dwBed:gpio8
axes_map: -y,x,z

[resonance_tester]
accel_chip_x: lis2dw hotend
accel_chip_y: lis2dw bed
probe_points: 147, 158, 20
#accel_chip: lis2dw 
---------------

Macro VIBRATIONS_CALIBRATION generates csv file names based on chip, so graph generation failed on line
globbed_files = glob.glob('/tmp/adxl345-*.csv')

Frix-x, 2023-11-21:
Thanks for the issue!

I didn't thought about making this name more flexible since I never thought about the other compatible chips. But thanks for pointing it!

However, since almost all people are running the good old ADXL345, this issue is not having a big impact for now. Moreover, looking at the datasheets of the chips, it look like the ADXL345 is more precise than the LIS2DW so ADXL345 is still a prefered chip. So I will add this to the backlog but it is not on my priority list since I need to work on other things before. If you want to do a PR in the meantime, I'll be happy to merge it :)

JurajMik, 2023-11-21:
Hi, please see
https://biqu.equipment/products/adxl-345-accelerometer-board-for-36-stepper-motors?variant=40446852759650
S2DW V1.0 has higher sensitivity, lower noise and improved temperature stability over ADXL345.

OoSTARTERoO, 2023-11-28:
Hi! I've recently got into this awesome project and happens to have the same problem.
Belts graph and input shaper graphs work, but Viberations graph doesn't
This is console log
```
오후 4:55
Command {plot_graph} finished

오후 4:55
No CSV files found in the /tmp folder to create the vibration graphs!

오후 4:55
Running Command {plot_graph}...:

오후 4:55
echo: Graphs generation... Please wait a minute or two and look in the configured folder.

오후 4:55
Writing raw accelerometer data to /tmp/lis2dw-sp200_00n1.csv file
```

Since the default option doesn't work, I changed ACCEL_CHIP to lis2dw
Which actually runs, and saves some data but eventually it cannot generate the graph.
[lis2dw-sp102_00n1.zip](https://github.com/Frix-x/klippain-shaketune/files/13485217/lis2dw-sp102_00n1.zip)
here's the raw data
Hope this helps!

Frix-x, 2023-11-29:
Hello, yes this is normal due to how the script currently handle the vibration files here: https://github.com/Frix-x/klippain-shaketune/blob/d1c18748d8235a038b713662f2ecb17873405352/K-ShakeTune/scripts/is_workflow.py#L123-L162

It's looking for files with the `adxl345` name inside. I'll see how to improve it and standardize the system

JurajMik, 2023-11-30:
Some inspiration (I did it like that...)

Frix-x, 2023-12-01:
Oh yes, this part is indeed a good idea!

> `RUN_SHELL_COMMAND CMD=plot_graph PARAMS="VIBRATIONS {direction} {accel_chip}"`

I'll use you suggestion for the next version!

JurajMik, 2023-12-14:
Even AXES_MAP_CALIBRATION macro is not working (you recommend replace adxl.. with our chip name), because also find_axesmap (is_workflow.py) has baked "adxl345" in
globbed_files = glob.glob('/tmp/adxl345-*.csv')
print("No CSV files found in the /tmp folder to analyze and find the axes_map!")
So I have csv in tmp folder, but as axpected, procedure fails with

SpeCter, 2023-12-24:
Just saw this after creating the same issue. Got the same problem and changing the fixed string to search for it works but the spectrogram looks kinda scuffed not sure what I forgot.

Frix-x, 2023-12-26:
> Just saw this after creating the same issue. Got the same problem and changing the fixed string to search for it works but the spectrogram looks kinda scuffed not sure what I forgot.

Can you share the spectrogram to have a look?
I need to fix some bugs and try to lower the memory use of the Shake&Tune package first and then I'll work on this issue.

SpeCter, 2023-12-26:
[image]

SpeCter, 2023-12-26:
[image]

SpeCter, 2023-12-26:
> Can you share the spectrogram to have a look? I need to fix some bugs and try to lower the memory use of the Shake&Tune package first and then I'll work on this issue.

[image]

Frix-x, 2024-01-07:
Hello, thanks for sharing all your data.

Quick update: I've worked on this and it's mostly finished (just need more testing).

I've done a major rewriting of how the parameters are passed in order to be more flexible. This allow to change the number of results to keep in the result folder, and also if the CSV file need to be archived or only the PNGs, or a couple of other things like the adxl name for the vibration graphs, etc... :)

I plan to release it this weekend alongside the code optimizations that were done for the memory issue #28 .

JurajMik, 2024-01-07:
Thank you for great work.  
Just quick-view workflow-rework branch -> is_workflow.py:

cleanfilename = os.path.basename(filename).replace('adxl345', f'vibr_{current_date}')

Frix-x, 2024-01-07:
> cleanfilename = os.path.basename(filename).replace('adxl345', f'vibr_{current_date}')

Good catch, thanks. I've fixed it in my last commit. Currently installing a LIS2DW accelerometer to test it and validate before merging :)

closed 2024-01-09 by Frix-x