Frix-x/klippain-shaketune #12: Allow to adjust paths to special Klipper installations

alienboyxp, 2023-11-23:
### Describe the feature or hardware support you'd like

In some cases new Klipper machines, yes brands like to be creative :), uses other paths for Klipper binaries and config like QIDI or K1. 
Should be interesting to had a global variable to adjust those paths used by all your great python scripts.

### Additional context or information

As an example for QIDI:
Klipper config -> /home/mks/klipper_config/
Klipper binaries -> /home/mks/klipper/
For Creality K1:
Klkipper config -> /usr/data/printer_data/
Klipper binaries ->

Frix-x, 2023-11-27:
Hello thanks for this request :)

I've looked a little bit at the problem and it seems that it will be pretty hard to make it work on Creality K1 printers since it need a root access, the OS is not debian based so the packages may be differents and I don't have the machine myself to test it ^^ But if someone have the machine and want to give it a try, I'm open to integrate it as a PR.

Regarding the QIDI machines, it should already work correctly since i'm always using the user path expansion variable `${HOME}` or `~/xxx`. So it should expand correctly to whatever user is set on the machine (pi, mks or biqu for example)

alienboyxp, 2023-11-27:
@Frix-x thanks for your reply!!!
On K1 machines it's a problem as the Klipper version, as the QIDI, are really old and not have the  libraries for some process... Having your solution with all need in one place was great but probably as you commented should be complex.
I had my K1 Max and I can test without problems.
For QIDI was implemented some basic scripts and macros to generate the graphs and works well, if you have time you can review here https://labs-3dwork-io.translate.goog/reviews-guias/impresoras-3d/fdm/qidi/qidi-x-max-3/qidi-x-max-3-technical-guide?_x_tr_sl=es&_x_tr_tl=en&_x_tr_hl=es&_x_tr_pto=wapp at Input Shaper section, but your method provide interested and extra information compared to standard. The problem on QIDI using your setup is an error related to Plot... I will try tomorrow and share the exact error... I do not investigate in detail but probably will be due QIDI old Klipper version.
Again thanks for answering and your excellent work!!!

Frix-x, 2023-11-29:
Yes sure, if you are equiped with the machines, please share the errors encountered, it will help me to try to make it more compatible :)

Frix-x, 2023-11-30:
> I get an error on a Qidi X Max-3 machine that plot is not generating after executing BELTS_SHAPER_CALIBRATION.
> 
> This is the error message: // Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file // Unknown command:"RESPOND" !! Error running command {plot_graph}
> 
> Really hope to get it run. Great piece of work, thank you very much!

This is unlikely the error: the RESPOND command is only used to gives some logs in the Mainsail/Fluidd console and it doesn't crash if not defined. To fix this you can just add `[respond]` somewhere in your config and it will solve this "unknown RESPOND command" warning.
However, I would like to get your klippy.log to see what is the reason of the plot_graph failure

ohyeah8000, 2023-11-30:
Bonjour!

Sure no problem. Here’s the slippy.log for your information.

Thank you and kind regards
Alex

ohyeah8000, 2023-11-30:
JFI: Now the error message change a bit after adding [RESPOND]….

alienboyxp, 2023-11-30:
As @ohyeah8000 same error
[image]
Error from the log:
Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file
```
shell_command: Command {plot_graph} failed
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/extras/gcode_shell_command.py", line 54, in cmd_RUN_SHELL_COMMAND
    self.command + gcode_params, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
Error running command {plot_graph}
Error running command {plot_graph}
```
This is the klipper/scripts directory from QIDI:
[image]

This is how I do myself currently for this machine:
```
#!/bin/bash
#set -e -u -o pipefail

## the TEST_RESONANCES command seems to return before all data is completely
## written. it'd be better to poll for the files to be completely written,
## but this has been reliable for me so far…
sleep 10

outdir=/home/mks/klipper_config/input_shaper
if [ ! -d "${outdir}" ]; then
    mkdir "${outdir}"
fi

/home/mks/klipper/scripts/graph_accelerometer.py \
    -c /tmp/*.csv \
    -o "${outdir}/belt-tension-resonances-$( date +'%Y-%m-%d-%H%M%S' ).png"

rm /tmp/raw_data_axis*
```

Finally the Klippy.log:
[klippy (1).log](https://github.com/Frix-x/klippain-shaketune/files/13515208/klippy.1.log)

Probably the system from QIDI miss to install some dependencies needed for your scripts.