Frix-x/klippain-shaketune #13: sometimes it works, sometimes i get an error in plot_graph

jpbnto, 2023-11-24:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v1.1.2-2-g7c76be50

### Describe the bug and expected behavior

04:10:03 PM
Re-enabled [input_shaper]
04:10:05 PM
Writing raw accelerometer data to /tmp/raw_data_y_y.csv file
04:10:05 PM
echo: Y axis shaper graphs generation...
04:10:05 PM
Running Command {plot_graph}...:
04:10:08 PM
Traceback (most recent call last):
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 232, in
04:10:08 PM
main()
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 218, in main
04:10:08 PM
fig, png_filename = get_shaper_graph()
File "/home/pi/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 117, in get_shaper_graph
fig = shaper_calibration([new_file], KLIPPER_FOLDER)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 332, in shaper_calibration
04:10:08 PM
datas = [parse_log(fn) for fn in lognames]
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 332, in
04:10:08 PM
datas = [parse_log(fn) for fn in lognames]
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 312, in parse_log
if not header.startswith('freq,psd_x,psd_y,psd_z,psd_xyz'):
UnboundLocalError: local variable 'header' referenced before assignment
04:10:08 PM
Command {plot_graph} finished

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/13456617/klippy.log)

Frix-x, 2023-11-24:
Thanks for your issue!

This is error happened in the past on low powered board or Pi with slow SD card filesystems. But it was solved... I think my last commit introduced a regression and I will fix it ASAP.

Frix-x, 2023-11-24:
If you update to the latest commit on the main branch, it should work correctly now as I reverted to the old checking mechanism

closed 2023-11-24 by Frix-x