Frix-x/klippain-shaketune #22: Belt shaper py errors out

sbtoonz, 2023-12-11:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

9ce3677

### Describe the bug and expected behavior

```
9:21 PM
Command {plot_graph} finished
9:21 PM
f12 = source_data[y_indices, x_indices + 1]
IndexError: index 365 is out of bounds for axis 1 with size 365
9:21 PM
combined_data, bins, t = combined_spectrogram(data1, data2)
File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 308, in combined_spectrogram
pdata2_interpolated = interpolate_2d(t1, bins1, t1, bins1, pdata2_aligned)
File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 237, in interpolate_2d
9:21 PM
main()
File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 230, in main
fig, png_filename = get_belts_graph()
File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 98, in get_belts_graph
fig = belts_calibration(lognames, KLIPPER_FOLDER)
File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 603, in belts_calibration
plot_difference_spectrogram(ax2, datas[0], datas[1], signal1, signal2, similarity_factor, max_freq)
File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 467, in plot_difference_spectrogram
9:21 PM
Belts estimated similarity: 85.5%
Traceback (most recent call last):
File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 246, in
```

The python seems to error out and not produce an image for me. 

Running on a CB1 on m5p

Expected behavior:
Results in image as per readme 
Experienced results:
error and no image

### Additional information and klippy.log

[Uploading klippy (31).log…]()

sbtoonz, 2023-12-11:
```
Belts estimated similarity: 85.5%
Traceback (most recent call last):
  File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 246, in <module>

    main()
  File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 230, in main
    fig, png_filename = get_belts_graph()
  File "/home/biqu/printer_data/config/K-ShakeTune/scripts/is_workflow.py", line 98, in get_belts_graph
    fig = belts_calibration(lognames, KLIPPER_FOLDER)
  File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 603, in belts_calibration
    plot_difference_spectrogram(ax2, datas[0], datas[1], signal1, signal2, similarity_factor, max_freq)
  File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 467, in plot_difference_spectrogram

    combined_data, bins, t = combined_spectrogram(data1, data2)
  File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 308, in combined_spectrogram
    pdata2_interpolated = interpolate_2d(t1, bins1, t1, bins1, pdata2_aligned)
  File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 237, in interpolate_2d

    f12 = source_data[y_indices, x_indices + 1]
IndexError: index 365 is out of bounds for axis 1 with size 365
```

Frix-x, 2023-12-11:
Hello, I just pushed a global update to v2.0.0 while you were writing this issue.

This new version had a lot of changes in the codebase and the error you are having is in some part of the code that doesn't exist anymore... So can you update and let me know if it's working or if you still get a bug somewhere?

sbtoonz, 2023-12-11:
Now the bug seems to be that klippain-shakentune is not respecting my numpy install 
```

raceback (most recent call last):
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/core/__init__.py", line 24, in <module>

    from . import multiarray
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/core/multiarray.py", line 10, in <module>

    from . import overrides
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/core/overrides.py", line 8, in <module>

    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/__init__.py", line 130, in <module>

    from numpy.__config__ import show as show_config
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/__config__.py", line 4, in <module>

    from numpy.core._multiarray_umath import (
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/core/__init__.py", line 50, in <module>

    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/home/zarboz/klippain_shaketune-env/bin/python"
  * The NumPy version is: "1.26.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/zarboz/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 30, in <module>
    from graph_belts import belts_calibration
  File "/home/zarboz/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 14, in <module>

    import optparse, matplotlib, sys, importlib, os
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/matplotlib/__init__.py", line 156, in <module>

Write g-code response
Traceback (most recent call last):
  File "/home/zarboz/klipper/klippy/gcode.py", line 459, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
BlockingIOError: [Errno 11] Resource temporarily unavailable
    import numpy
  File "/home/zarboz/klippain_shaketune-env/lib/python3.9/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

Command {shaketune} finished
```

Frix-x, 2023-12-12:
Oh, it look likes you are missing some system dependencies. Just install them like this and it should solve your error. I'll add these dependencies to the install script to avoid this in the future :)
```bash
sudo apt update
sudo apt install libopenblas-dev libatlas-base-dev -y
```

sbtoonz, 2023-12-12:
Working as expected now

closed 2023-12-12 by sbtoonz

Yonn-Trimoreau, 2023-12-16:
Hi @sbtoonz sorry to bother you here. We all need your help on itemdrawers, can you please check it?