Frix-x/klippain-shaketune #25: Error running command {shaketune}

Dejv1cz, 2023-12-13:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0

### Describe the bug and expected behavior

Everything starts up fine
csv files are created and written
but then it says: Error running command {shaketune}

Frix-x, 2023-12-14:
Can you add more info like the log and what is displayed in the console? It will be pretty hard for me to debug it without these information

Mike-Gough, 2023-12-14:
I had the same issue and eventually worked it out. What seemed to fix it for me is the installer put everything in `~/printer_data/config/scripts/K-ShakeTune` and after I moved it to the parent folder `~/printer_data/config/K-ShakeTune` I no longer got the error

Dejv1cz, 2023-12-15:
I tried to remove and reinstall everything and suddenly it works

Frix-x, 2023-12-15:
Oh ok, this is strange. I'll have a look to see how that's possible... But I'll close this issue if everything is running fine for you now :)

closed 2023-12-15 by Frix-x

dragoonmc, 2024-04-18:
I had a similar experience. might be from me adding the K-ShakeTune folder to the config folder while installing. also was missing the scripts in Klippain folder.
I am on a CB1 instead of a Pi if that makes any difference