Frix-x/klippain-shaketune #26: ModuleNotFoundError: No module named 'scipy'

Hr46ph, 2023-12-14:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0-0-gc7e39da5

### Describe the bug and expected behavior

I am not sure which version I was running (I installed recently in the past couple of weeks).
I am on RatOS 2, fully updated, and installed Klippain following the instructions a couple of weeks ago. Its been working flawlessly.

After the update today I get:
```
16:43 Command {shaketune} finished
16:43 /home/pi/printer_data/config/K-ShakeTune/scripts/shaketune.sh: line 5: deactivate: command not found
16:43 import scipy
ModuleNotFoundError: No module named 'scipy'
16:43 from graph_belts import belts_calibration
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_belts.py", line 17, in
16:43 Traceback (most recent call last):
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 30, in
16:43 /home/pi/printer_data/config/K-ShakeTune/scripts/shaketune.sh: line 3: /home/pi/klippain_shaketune-env/bin/activate: No such file or directory
16:43 Running Command {shaketune}...:
```

### Additional information and klippy.log

I assume its missing a python module scipy.
```
$ pi@vcore-3 ~ $ apt search scipy
python3-scipy/oldstable 1.6.0-2 arm64
  scientific tools for Python 3
```

But since you also switched to a venv I'm going to hold off trying to fix it myself and wait for your reply.

Klippy attached. If you need more info just ask.
[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/13675502/klippy.log)

stevereno30, 2023-12-14:
I also have this issue. I was using version 1.2.0 of Shake & Tune without issue up until version 2.0.0 was released. I was part of the array value to big issue that was fixed during the 2.0.0 release, but I had solved that particular issue by downgrading PiOS to the legacy distro. I will try upgrading PiOS to see if it fixes this new scipi issue.

Hr46ph, 2023-12-15:
`line 5: deactivate: command not found`

is a venv error. Perhaps the venv didn't get activated or something.

Frix-x, 2023-12-15:
Yes, you are running into the breaking change mentionned in [the release notes of the v2.0.0](https://github.com/Frix-x/klippain-shaketune/releases/tag/v2.0.0). Just run the steps here and everything should work :)

Also if you are running Debian Buster (or an older system that has only Python 3.7 as the last Python version), you may need to stay on v1.x for now since this Scipy requirement apparently need to also have at least Python 3.8... I'm currently looking to mitigate this issue in order to allow older OS running Shake&Tune.

Hr46ph, 2023-12-16:
Well, that doesn't install `python3-venv`, which apparently was missing on Debian GNU/Linux 11 (bullseye).

Also, I would kindly advise you to not instruct users to use `sudo` unless it absolutely required. There is no need for it on the rm command as there are no files owned other than `pi`. As you may know, sudo is dangerous and a mistake is easily made.

In any case, your work on this is highly appreciated! Thank you!

closed 2023-12-16 by Hr46ph

leitenky, 2023-12-18:
I can confirm running the setup again fixed my issue.