Frix-x/klippain-shaketune #28: Memory Issue(?) Process was terminated while running the script

SHKinsem, 2023-12-17:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0-2-g2a627a1f

### Describe the bug and expected behavior

I was running the gcode BELTS_SHAPER_CALIBRATION, and sometimes it would pop out something shown below:

```
// Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file
echo: Belts comparative frequency profile generation...
echo: This may take some time (3-5min)
// Running Command {shaketune}...:
// /home/klipper/printer_data/config/K-ShakeTune/scripts/shaketune.sh: line 4: 120654 Killed python ~/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py "$@"
// Command {shaketune} finished
```

And sometimes the klipper would restart without any warning.

The other commands can be run flawlessly. 
I am using orangepi zero3 with 1gb of ram.

SHKinsem, 2023-12-17:
After a few tests, I stopped homeassistant and it turned out working... But however, is there anything could be done to reduce the ram usage?

Frix-x, 2023-12-18:
> After a few tests, I stopped homeassistant and it turned out working... But however, is there anything could be done to reduce the ram usage?

Oh ok... I already optimized the RAM usage a lot in the v2.x release and I don't think I can do much more easily... Total RAM usage is around 150MB on my machine. I'll try to monitor it closely and let you know if I find out something...

Phil1988, 2023-12-19:
Total RAM usage on my machone is ~350MB only running OS + klipper.
When running `BELTS_SHAPER_CALIBRATION` it eats the remaining 650MB and moonraker restarts:
[image]

Could you maybe check again if you can do anything about it?

Belt calibration is pretty handy to tune the 3D printer :)

Phil1988, 2023-12-22:
I reinstalled my whole OS and tested it.

It works now and I got this graph:
[image]

Is it normal that the table line "Peaks B" is missing or did something fail here?

konkers, 2023-12-22:
I just ran into this too. `is_workflow.py` runs for 10-20s then there's a large memory spike and the kernel OOM killer fires off.  I've attached my CSV files in case this is data dependent
[belts.zip](https://github.com/Frix-x/klippain-shaketune/files/13755341/belts.zip)
.

konkers, 2023-12-22:
Here's the list of python package version in my venv: ```$ ./klippain_shaketune-env/bin/pip list
Package         Version
--------------- -------
contourpy       1.2.0
cycler          0.12.1
fonttools       4.45.1
kiwisolver      1.4.5
matplotlib      3.8.2
numpy           1.26.2
packaging       23.2
Pillow          10.1.0
pip             23.3.2
pyparsing       3.1.1
python-dateutil 2.8.2
scipy           1.11.4
setuptools      66.1.1
six             1.16.0```

Frix-x, 2023-12-26:
Thanks for sharing your data. This issue will be my priority starting from today and I hope to be able to lower the memory usage a bit more...

> Is it normal that the table line "Peaks B" is missing or did something fail here?

On your graph you only have one pair of peaks (A1 paired with A2), so that's why there is only one line in the table for the peaks A. On the other hand, the Peak 1 is detected as alone and not paired so I can't compute any stats about it.

Frix-x, 2024-01-02:
I've worked on this issue a lot for a couple of days by doing a lot of memory tracing and analysis of my code.

So I've found some memory culprits in two matplotlib graphing functions. So I've done some refactoring of the code and implemented some workarounds for this on my side. This allowed to reduce the total memory consumption by ~75% and speed up the graph generation around 15x on my side for the belt graph!

I just need to clean up a little bit and reproduce the enhancements for the input shaper graphs and vibrations graphs before issuing a new release. Stay tuned it will come soon :)

Phil1988, 2024-01-03:
Thats an amazing result!
Thanks that you took the time to trace it back and to solve this issue - its also nice that the speed also could be increased by this 👍

MagicBugsBunny, 2024-01-04:
When will you release the new Version?
I can't run any InputShaper on my Zero 2w with ShakeTune :-(

Frix-x, 2024-01-04:
You can already try the develop version that was pushed just a minute ago with all the changes. I plan to release it to main very soon but need to test and polish it a little bit to be sure everything is alright.
Also some part of the documentation needs to be updated regarding the vibration graph that was updated with a new element :)

[image]

MagicBugsBunny, 2024-01-05:
Develop version is working for me

Frix-x, 2024-01-05:
> Develop version is working for me

Can you confirm that it's working on a PiZero 2W? But thanks for the feedback, that's great news :)

MagicBugsBunny, 2024-01-05:
Yes, have done many IS and Belt Tests at the Moment

closed 2024-01-09 by Frix-x

konkers, 2024-01-09:
Sorry I never got back to your question, I've been out sick for the last week or so.  Thanks so much for digging deep into this!