Frix-x/klippain-shaketune #29: green zone confusino

Hr46ph, 2023-12-17:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0-0-gc7e39da5

### Describe the bug and expected behavior

So both graphs have been taken on the same machine, with absolutely no changes, except for 1 thing: the steppers direction have been inverted by reversing the wires, and the pin rotation in Klipper marked with a `!` to correct the reverse wiring.
The first graph is with reverse wiring. The second with the default wiring.

When we look at the green zones on each, one might think the second graph is better but I doubt that is the case (hence I am opening the report).

The green zone on the second graph goes up to 0.4 on a 1e8 scale. 40,000,000 to be exact. This is equivalent to 4.0 on a 1e7 scale, the end of 400mm/s. Basically, the entire first graph should be green zone.

Unless I misunderstand something, please clarify then if its not too much to ask 😸.
 
[image]

[image]

Thanks for your time!

3dcase, 2023-12-17:
I see it as follows:
First of all you need to calculate the total amount of energy in those vibrations.
I take it simply, may not be perfect but will serve the purpose.
Max value of peak and then take the percentage of that zone. In the top graph this is 1.8*10^7=18000000
26.54%=4777200 of whatever units is in the graph.
In the bottom graph there are 5 zones.
1st zone is already 3504000
2nd zone is 6660000
3rd zone is 7892000
4th and 5th zones are even bigger. When you add them up you get a far bigger amount of energy that goes into the vibration.
Apart from the fact you have only 1 zone in the first graph, the total energy is far far less as well so the top graph is way better I think.
Now I am in no way an expert in these graphs but I merely follow some logic when looking at the values displayed in the graphs and this conclusion is what I made out of it.
I am also interesting to see what others have to say about it, always up for learning something new.

Frix-x, 2023-12-18:
Hello,

You are indeed right that the first graph looks better. In fact, the right side of these two graphs are quite similar and the changes are the spikes from the left side that are higher on the second graph.

Regarding the green zones, they are in fact not calculated using a "constant" threshold. The goal is to take all kind of machines into account and guide the user to the speed ranges he can use based on his own graph. Here is how I calculate these zones:
1. First, I calculate the mean and standard deviation of the entire signal. This is used to establish a baseline for what will be the "low energy zones".
2. Then I compute a dynamic threshold (valid only for this signal) as the mean value minus 0.1 * standard deviation. This allows this threshold to depend on the signal's height, but also to account for a very noisy signal to avoid selecting small peaks. This 0.1 value was found experimentally on a large set of graphs to find something that works pretty well for everyone.
3. The low energy zones are then identified on the graph by looking at the parts below this threshold.
4. Finally, the low energy zones are also sorted by calculating their mean value and comparing it to the mean energy of the entire signal calculated at step 1. This comparaison is expressed as a percentage in the upper legend for each zone.

closed 2024-04-30 by Frix-x