Frix-x/klippain-shaketune #31: Calibration fails

Phil1988, 2023-12-18:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0-2-g2a627a1f

### Describe the bug and expected behavior

When using BELTS_SHAPER_CALIBRATION it fails on generating the graphs with this message in the console:
`/home/mks/printer_data/config/K-ShakeTune/scripts/shaketune.sh: line 4: 20715 Killed python ~/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py "$@"`
[image]

I tried it multiple times with different debian releases and the csv files being stored here but no images of the plottet graphs:
[image]

I used these parameters for the belt calibration:
[image]

AXES_SHAPER_CALIBRATION works with no issues.

What can I do to check where it fails and improve this great extension?

Frix-x, 2023-12-18:
This is a duplicate of #28 and is likely due to a too low RAM available when running the script. I already optimized it quite a lot in the V2.0.0 version but apparently not enough... If you have other processes running on your Pi, try to stop them to free up some RAM before running the belt macro (which is the one that consume the more RAM).

On #28, the user is running Home Assistant on its Pi and this apparently solved the issue but I agree that it's not perfect and I'll need to look at it. I'll close this issue in favor of the other one :)

closed 2023-12-18 by Frix-x

Phil1988, 2023-12-18:
There is only klipper (+moonraker + mainsail) and your script running.
Nothing more, but my RK3328 also only has 1GB RAM.

Is there any workaround, that I can use?