Frix-x/klippain-shaketune #32: Installation Fails, No Matching contourpy?

TjardF, 2023-12-18:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

Current

### Describe the bug and expected behavior

I tried installing like given in the manual, sadly the install fails.
Any tips?

[image]

Frix-x, 2023-12-19:
This is because you are running an old system OS that has only Python3.7 available in its repo. **The Shake&Tune package need at least Python3.9** (this is what you can read in the error you posted above).

So there is currently two options for you:
1. You update Python to >=3.9 manually (3.10 or 3.11 is a good choice since they are the versions currently supported but 3.12 should also work correctly)
2. You update your OS to something more recent like Debian Bulleye or Bookworm that both have more recent Python versions included by default. If you backup your printer configuration and install the last MainsailOS or official RpiOS release, you should be good to go.

On my side, this is a new requirement that I did not expected when switching to Scipy for the computation of the spectrogram and I'm currently looking at optimizing the code to lower it and allow older Python3.7 installs to work, but not sure I will be able to do it...

emintjeee, 2023-12-24:
i have the same, installed python 3.12 but it didnt resolved the issue.

emintjeee, 2023-12-26:
As discussed on Discord with @Frix-x  re-installed my PI4 with Bookworm instead of Buster.
After this it was resolved! @TjardF

TjardF, 2023-12-27:
Oh sorry.
I forgot to reply.
With reinstalling new mainsail os it did its thing.
The System was about 18months old…

closed 2023-12-27 by TjardF