Frix-x/klippain-shaketune #36: need to add dependancy "Matplotlib"

PaulMellor, 2023-12-31:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.0.0-2-g2a627a1f

### Describe the bug and expected behavior

After running the Axes shaper calibration I was getting the following error. as found that the matplotlib and lumpy was not installed suggest adding them to the dependance check list.

Traceback (most recent call last):
File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 43, in
from graph_belts import belts_calibration
File "/home/pi/klippain_config/scripts/is_workflow/graph_belts.py", line 21, in
import optparse, matplotlib, sys, importlib, os
ModuleNotFoundError: No module named 'matplotlib'

locki-cz, 2024-01-01:
uninstall and install it again, it help for me

Frix-x, 2024-01-02:
Are you using the last version?
I'm asking as since the v2.x of Shake&Tune, everything run in a Python virtual environment allowing me to manage the dependencies (and matplotlib is indeed installed automatically in this venv).

closed 2024-01-05 by Frix-x