Frix-x/klippain-shaketune #37: Beacon RevH include

spotless20, 2024-01-05:
### Describe the feature or hardware support you'd like

Hey, maybe just a skill issue of myself. 

Got a beacon REV H with "
[resonance_tester]
accel_chip: beacon
"

And its blaming me that its not an adxl345 

Is there any option that you could include that the beacon is also working 

Thank you :)

NotTeoZz, 2024-01-07:
Quote, got an idm scanner with the lis2dw accelerometer and it does not work even using ACCEL_CHIP=lis2dw

Frix-x, 2024-01-07:
I recently learned that the new beacon includes an onboard accelerometer, which seems like a smart addition. I'm considering purchasing one.

Looking at its documentation, it appears that the following configuration is all that's needed to enable the accelerometer feature in the beacon (provided that the latest .py plugin was installed):
```
[resonance_tester]
accel_chip: beacon
```
This setup should enable the use of standard accelerometer commands in Klipper, such as MEASURE_AXES_NOISE. If you encounter issues with Klipper not restarting properly or if the MEASURE_AXES_NOISE command isn’t working, I recommend referring back to the beacon's documentation to ensure the installation instructions were followed correctly.

On Shake&Tune side, if the accelerometer commands are working correctly, then both belt and shaper graphs should work seamlessly. However, please note, only the ADXL345 accelerometer is supported for the vibration graphs for now but I plan to extend support to other accelerometer chips very soon in the upcoming version.

Frix-x, 2024-01-07:
See #11

closed 2024-01-07 by Frix-x