Frix-x/klippain-shaketune #40: IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

PerLangkjaer, 2024-01-09:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-1-gac96cb2e

### Describe the bug and expected behavior

I get this issue when I run the 

calibration_data = self.background_process_exec(
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
return method(*args)
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 161, in calc_freq_response
T = data[-1,0] - data[0,0]
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

Frix-x, 2024-01-11:
Hello,

Is this error happenning everytime you run the script? Can you give me the problematic CSV files please to have a look?
Also please reboot the Pi (to flush the /tmp folder) and let me know if it solve the issue.

PerLangkjaer, 2024-01-12:
As your question in your comment, I had not rebooted the PI.
After doing a complete power-cycle everything works!!

Frix-x, 2024-01-15:
Ok, this may be due to some csv artifacts from older runs before Shake&Tune was installed. I'll close the issue for now but feel free to reopen it if it come back.