Frix-x/klippain-shaketune #42: GitPython module missing

Zeanon, 2024-01-11:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-0-g84c406b4

### Describe the bug and expected behavior

When trying to do the belt graphing, I was presented with this error:
`from git import GitCommandError, Repo ModuleNotFoundError: No module named 'git'`
Fix was easy enough, just install GitPython to the virtual env but I thought I'd give you a heads up

Frix-x, 2024-01-11:
This is because you updated from a previous version of S&T. In this case there is indeed a breaking change due to the way moonraker manage the install_script entry that I was not aware.

Installing manually the GitPython module will make it work today but you will get the same problem on the next update until you update the moonraker section as decribed in the release notes here: https://github.com/Frix-x/klippain-shaketune/releases/tag/v2.5.0

Zeanon, 2024-01-11:
Oh my bad, didn't read the changelog 
Sorry for bothering you about it XD

Frix-x, 2024-01-11:
Haha no problem. I'll close the issue then :)

closed 2024-01-11 by Frix-x