Frix-x/klippain-shaketune #43: ERROR: No matching distribution found for matplotlib==3.8.2

HydroRubi16, 2024-01-13:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

2.5.0

### Describe the bug and expected behavior

Hello,

When I attempt to install the Shake&Tune package via SSH I get the following errors:

ERROR: Ignored the following versions that require a different python version: 3.6.0 Requires-Python >=3.8; 3.6.0rc1 Requires-Python >=3.8; 3.6.0rc2 Requires-Python >=3.8; 3.6.1 Requires-Python >=3.8; 3.6.2 Requires-Python >=3.8; 3.6.3 Requires-Python >=3.8; 3.7.0 Requires-Python >=3.8; 3.7.0rc1 Requires-Python >=3.8; 3.7.1 Requires-Python >=3.8; 3.7.2 Requires-Python >=3.8; 3.7.3 Requires-Python >=3.8; 3.7.4 Requires-Python >=3.8; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 3.8.2 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement matplotlib==3.8.2 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2, 3.5.3)
ERROR: No matching distribution found for matplotlib==3.8.2

[image]

I completed all updates in Mainsail and attempted the install again - I get the same error.

Frix-x, 2024-01-14:
Hello,
This is because you are running an old system OS that has only Python3.7 available in its repos. **Shake&Tune needs at least Python3.9** (this is what you can read in the error you posted above). You can check this by running over SSH `python3 --version`.

So there is currently two options for you:

- You update Python to >=3.9 manually (3.10 or 3.11 are good choices since they are the versions currently supported but 3.12+ should also work correctly)
- But the easiest on a Pi is to update your OS to something more recent like Debian Bulleye or Bookworm that both have more recent Python versions included by default. If you backup your printer configuration and install the last MainsailOS or official RpiOS release with KIAUH, you should be good to go in no time.

HydroRubi16, 2024-01-17:
Thanks for the quick response! I’m going to do a fresh Mainsail install on a new MicroSD card, then import my current CFG files. I’ll report my findings. Thanks!

HydroRubi16, 2024-01-18:
That fixed my issue - I reloaded Mainsail and all is good now - Thanks!

closed 2024-01-18 by HydroRubi16