Frix-x/klippain-shaketune #44: No CSV files found in the /tmp folder to create the vibration graphs!

taganaka75, 2024-01-14:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-2-g0696a60b

### Describe the bug and expected behavior

I've a printer with RatOS installed,
When i run VIBRATIONS_CALIBRATION command i recieve the error "Invalid adxl345 id (got ff vs e5).

If i run VIBRATIONS_CALIBRATION ACCEL_CHIP="toolboard" or VIBRATIONS_CALIBRATION ACCEL_CHIP=toolboard
the commad start correctly and execute the data log but after:
"Machine and motors vibration graph generation..."
"This may take some time (3-5min)
I recieve the message "No CSV files found in the /tmp folder to create the vibration graphs!"

In the /tmp directory there are the .csv files with this name:
"adxl345-toolboard-sp20_00n1.csv" ecc.

What's i'm doing wrong?

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/13931271/klippy.log)

taganaka75, 2024-01-14:
If I rename "adxl345-toolboard-sp20_00n1.csv" to "toolboard-sp20_00n1.csv"the files in the /tmp folder, the program works and generate the plot...

Frix-x, 2024-01-14:
Hello,
I don't think you are doing something wrong but currently custom named accelerometer doesn't work for the vibration graphs.

So if you have something like `[adxl345 whatevername]`, this will not work. Shake&Tune is for now compatible only with vanilla `[adxl345]` or `[lis2dw]`, etc... without any custom name. But I already have it in the backlog and do plan on working on this subject soon.

The-Original-Reth, 2024-01-15:
This is my attempt to fix this. 
What it does it is checks to see if adxl_y is the same as adxl_x if true then it just does normal operation.
if adxl_y is not the same as adxl_x then it will run X with adxl_x and Y with adxl_y

[K-SnT_vibrations - seperate adxl for x and y.txt](https://github.com/Frix-x/klippain-shaketune/files/13941211/K-SnT_vibrations.-.seperate.adxl.for.x.and.y.txt)

The-Original-Reth, 2024-01-15:
will this work for lis2dw also, or is adxl fixed in there

ndanyluk, 2024-03-06:
Hey @Frix-x thanks for this awesome module! Currently this issue also affects bed slingers as well since you need 2 adxls by necessity (one for toolhead and one for the bed).

I'll take a look at the code and see if I can at least fix this for bedflingers (which we could enforce a set "toolhead" and "bed" naming scheme to have it follow a convention S&T can follow

Frix-x, 2024-03-06:
> Hey @Frix-x thanks for this awesome module! Currently this issue also affects bed slingers as well since you need 2 adxls by necessity (one for toolhead and one for the bed).
>
> I'll take a look at [...]

Thanks for the proposal.
I have a plan already but need to find time to work on it. But to sum up, I want to retrieve the acelerometer type from the Klipper macro by using jinja and looking at the "printer" object in order to send it to the Python script and replace the hardcoded "adxl345" string with this. If you find out how to do this, it would be wonderful and I'll be happy to accept a PR :)

Frix-x, 2024-05-19:
Just to let you know that the new version is almost finished. And in this new version, I've made a fix for this problem.

Now S&T runs as a real Klipper extras plugin: so I've got access to the accelerometer objects directly in the Python side, and I've added some logic to use the best accelerometer each time, based on the ones configured in the `[resonance_tester]` config section. So this should no longer be a problem!

closed 2024-06-13 by Frix-x