Frix-x/klippain-shaketune #46: Cannot write to '/home/pi/klipper/klippy/extras/gcode_shell_command.py'

Lekkz, 2024-01-14:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-3-geab10ce5

### Describe the bug and expected behavior

when I installed the newest version i keep getting this error on the installation screen
"Cannot write to '/home/pi/klipper/klippy/extras/gcode_shell_command.py' (No such file or directory)."

also which might be related as soon I append the cfg file I get an error on Mainsail saying
"Section 'gcode_shell_command shaketune' is not a valid config section"
[klippy (18).log](https://github.com/Frix-x/klippain-shaketune/files/13933317/klippy.18.log)

Frix-x, 2024-01-17:
That's strange. Is this a standard OS?
You may try to install the plugin manually using KIAUH in the advanced section and see if this help?

Lekkz, 2024-01-18:
ok will try that.. as far as I know it is standard OS

HydroRubi16, 2024-01-18:
If it helps, I just ran into this very same issue.

SSH into your PI and go to the directory:
/home/pi/klipper/klippy/extras/

Then delete the current gcode_shell_command.py file (mine appeared to be a link, not an actual file).

Once deleted, reinstall ShakeTune.

Fixed.

Lekkz, 2024-01-24:
I can confirm @HydroRubi16 fix worked. sorry for the late response

closed 2024-01-24 by Lekkz