Frix-x/klippain-shaketune #47: Idea: put default amount of keep_results into a variables file

Zeanon, 2024-01-15:
### Describe the feature or hardware support you'd like

I think it would be nice, to be able to alter the default amount of keep_results without having to specify it in the command every time.
Solution would be rather easy, have one variable file, that gets copied instead of symlinked and the install script checks whether it already exists (if not, it gets copied, if it exists, it wont)
(I can do a pr for it, just wanted to check whether you would even consider it first)

Zeanon, 2024-01-15:
> Hi, thanks for the suggestion, and yes this is something already in my backlog. It was just not the priority but I definitely want to add this.
> 
> I'm really open for a PR if you want to work on this :) Also, at the same time I wanted to have the accelerometer chip name and the "keep_csv" in this variable in the same way. And maybe define a main read-only file as a simlink with default value that one could copy paste and override in his config if needed. What do you think about this? It would make it easier for updates and avoid potential breaking changes.

Sounds good, havent thought about those two as well but makes sense to also make them "static"
I would also add a version variable to it which gets checked on startup, to prevent mishaps (like I am doing here: https://github.com/LynxCrew/Klicky/blob/main/Variables/homing_variables.cfg)
I'll get to it later and write the code for it

Zeanon, 2024-01-15:
PR is out

Surion79, 2024-02-13:
could you please link your PR to the issue or vice versa? Thanks :)

Frix-x, 2024-02-14:
@Surion79 #48

closed 2024-06-13 by Frix-x