Frix-x/klippain-shaketune #49: How to install on a sonic pad?

jakyhaky, 2024-01-16:
### Describe the feature or hardware support you'd like

I'm using a sonic pad and log in putty with root account. And when I try to install it, I get an error message that I can't install with root account

Frix-x, 2024-01-17:
Hello,

I'm not familiar with the sonic pad but I heard that the system is locked and don't use standard Klipper/moonraker. If this is true, I'm not sure that you will be able to install S&T on it.

Is there any standard user available (beside root)?
What are the user config path? Are there the standard `~/printer_data/config`?

jakyhaky, 2024-01-18:
Are there any standard users available (other than the root)?
  - No, I can only use the root account
What is the user setup path? Is there a standard?
 - I don't know, but the printer.cfg file exists.

OctoEverywhere was installed successfully using putty.

henrivdr, 2024-04-12:
The Nebula pad (Retail, CR10 SE and Ender KE) has the same issue.
I guess now that they have official rooting support, we can create another user.  Like Creality should have done....

Frix-x, 2024-05-19:
Hello,

I've got some news. I have been working a lot on Shake&Tune the last few weeks and have converted all the code into a real Klipper plugin. Now the user limit should not be a problem anymore as it runs directly in the Klipper environment.

The release is not finished yet, I just have to polish a few things, but it should be out in the next days :)

sleewok, 2024-06-05:
Any update on this? I am running a rooted K1C and have a similar issue. It would be helpful if there were instructions to manually install your plugin.

Frix-x, 2024-06-09:
Sorry, I'm a bit late on the release as I started to add new tools to Shake&Tune haha
But I've finished most of the work now and I'm onto documentation now so it should arrive soon.

sleewok, 2024-06-10:
Looking forward to it! Thanks for putting the time into it.

Frix-x, 2024-06-14:
v4 was released a couple of hours ago. Can you tell me if this is working now?

jakyhaky, 2024-06-16:
login as: root
root@192.168.45.235's password:

BusyBox v1.27.2 () built-in shell (ash)

 ----------------------------------------------
 Linux (Neptune, 5C1C9C53)
 ----------------------------------------------
root@spad-2524:~# wget -O - https://raw.githubusercontent.com/Frix-x/klippain-sh
aketune/main/install.sh | bash
Connecting to raw.githubusercontent.com (185.199.110.133:443)
-                    100% |*******************************|  4863   0:00:00 ETA

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] This script must not be run as root!
root@spad-2524:~#

KennyVB, 2024-08-14:
hmm can't install it on rooted newest versiomn of ender KE???
```
 wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh --no-chec
k-certificate | bash
-sh: bash: not found
Connecting to raw.githubusercontent.com (185.199.108.133:443)
writing to stdout
```

it has the guroulz guy script installed entware etc. and guppy which all ready has a simple or what he thinks we need version sadly :( 

and since you said klipper enviroment how to install that thing? i had to use no-check apprently but nothing happens here

jakyhaky, 2025-06-03:
> [@jakyhaky](https://github.com/jakyhaky) hey, why do you close this issue? Root check still persists in install.sh script