Frix-x/klippain-shaketune #51: Installation issue "Someone may have tampered with them"

fsaad1984, 2024-01-19:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

V2.5.0

### Describe the bug and expected behavior

when running the install script, the installation fails with the following error:

ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    scipy==1.11.4 from https://www.piwheels.org/simple/scipy/scipy-1.11.4-cp39-cp39-linux_armv7l.whl#sha256=f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739 (from -r /home/pi/klippain_shaketune/requirements.txt (line 4)):
        Expected sha256 f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739
             Got        92b2e6ce020a771b5ed2517d0521b2b1cf60f4c22e1ce6d0a5b6ec57d3217ca9

Frix-x, 2024-01-19:
You may have a pip cache problem on your Pi. This can happens when you don't have a stable WiFi connection for example.
Try running this to clean the venv cache:
```bash
source ~/klippain_shaketune-env/bin/activate
python -m pip cache purge
deactivate
```
Then rerun the install script.

fsaad1984, 2024-01-19:
that indeed worked! Thank you :)

closed 2024-01-19 by Frix-x