Frix-x/klippain-shaketune #56: No CSV files found in the /tmp folder to create the vibration graphs!

wiggywoosaa, 2024-01-29:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-0-g84c406b4

### Describe the bug and expected behavior

After running Vibrations Calibrations or Axes Map Calibration I received the No CSV files found in the /tmp folder to create the vibration graphs! in the cli windows. I have checked the /tmp file and the files are indeed present

dhruvinsh, 2024-02-03:
Patch
```
diff --git a/K-ShakeTune/scripts/is_workflow.py b/K-ShakeTune/scripts/is_workflow.py
index dde04f6..1ea631c 100755
--- a/K-ShakeTune/scripts/is_workflow.py
+++ b/K-ShakeTune/scripts/is_workflow.py
@@ -136,7 +136,7 @@ def create_vibrations_graph(axis_name, accel, chip_name, keep_csv):
     current_date = datetime.now().strftime('%Y%m%d_%H%M%S')
     lognames = []
-    globbed_files = glob.glob(f'/tmp/{chip_name}-*.csv')
+    globbed_files = glob.glob(f'/tmp/adxl345-{chip_name}-*.csv')
     if not globbed_files:
         print("No CSV files found in the /tmp folder to create the vibration graphs!")
         sys.exit(1)
```

arsonists, 2024-02-09:
Good morning,

I have the same problem, how can I apply the update?

wiggywoosaa, 2024-02-09:
find this file -  K-ShakeTune/scripts/is_workflow.py

remove this line - globbed_files = glob.glob(f'/tmp/{chip_name}-*.csv')
replace it with this - globbed_files = glob.glob(f'/tmp/adxl345-{chip_name}-*.csv')

arsonists, 2024-02-09:
thank you very much

Frix-x, 2024-02-09:
> Patch
>
> ```
> diff --git a/K-ShakeTune/scripts/is_workflow.py b/K-ShakeTune/scripts/is_workflow.py
> index dde04f6..1ea631c 100755
> --- a/K-ShakeTune/scripts/is_workflow.py
> +++ [...]

Hello, thanks for sharing this patch. In fact it was like this before and I had to change it since it break the compatibility with LIS2DW and other kind of sensors. This issue is in fact a duplicate of #44 .

I'll work on a proper fix to be able to get it work with all kind of sensors and with configuration where they have "custom names". In the meantime, avoid using `[adxl345 my_custom_name]` for everything to work correctly :)

Frix-x, 2024-03-04:
I'll close this one in favor of #44

closed 2024-03-04 by Frix-x