Frix-x/klippain-shaketune #57: Install Error - Python & matplot version

DomLuu, 2024-01-31:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0

### Describe the bug and expected behavior

I got this Error: ERROR: Ignored the following versions that require a different python version: 3.6.0 Requires-Python >=3.8; 3.6.0rc1 Requires-Python >=3.8; 3.6.0rc2 Requires-Python >=3.8; 3.6.1 Requires-Python >=3.8; 3.6.2 Requires-Python >=3.8; 3.6.3 Requires-Python >=3.8; 3.7.0 Requires-Python >=3.8; 3.7.0rc1 Requires-Python >=3.8; 3.7.1 Requires-Python >=3.8; 3.7.2 Requires-Python >=3.8; 3.7.3 Requires-Python >=3.8; 3.7.4 Requires-Python >=3.8; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 3.8.2 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement matplotlib==3.8.2 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2, 3.5.3)
ERROR: No matching distribution found for matplotlib==3.8.2

Current Version: 
pi@mainsailos:~ $ python3.8 --version
Python 3.8.10

pi@mainsailos:~ $ python3 -m pip show matplotlib
Name: matplotlib
Version: 3.3.4

thx Domluu

DomLuu, 2024-01-31:
pi@mainsailos:~/klippain_shaketune $ ./install.sh

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

python3-venv is already installed
libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

[SETUP] Virtual environment already exists. Continuing...
[SETUP] Installing/Updating K-Shake&Tune dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /home/pi/klippain_shaketune-env/lib/python3.7/site-packages (23.3.2)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting GitPython==3.1.40 (from -r /home/pi/klippain_shaketune/requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/gitpython/GitPython-3.1.40-py3-none-any.whl (190 kB)
ERROR: Ignored the following versions that require a different python version: 3.6.0 Requires-Python >=3.8; 3.6.0rc1 Requires-Python >=3.8; 3.6.0rc2 Requires-Python >=3.8; 3.6.1 Requires-Python >=3.8; 3.6.2 Requires-Python >=3.8; 3.6.3 Requires-Python >=3.8; 3.7.0 Requires-Python >=3.8; 3.7.0rc1 Requires-Python >=3.8; 3.7.1 Requires-Python >=3.8; 3.7.2 Requires-Python >=3.8; 3.7.3 Requires-Python >=3.8; 3.7.4 Requires-Python >=3.8; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 3.8.2 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement matplotlib==3.8.2 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2, 3.5.3)
ERROR: No matching distribution found for matplotlib==3.8.2
pi@mainsailos:~/klippain_shaketune $

DomLuu, 2024-01-31:
The problem could be solved by adjusting the requirements.txt:
 
GitPython==3.1.40
matplotlib==3.3.4
numpy==1.19.5
scipy==1.7.3

Frix-x, 2024-02-01:
Shake&Tune is only compatible with Python 3.9 or later, which is a requirement for Scipy, Numpy and GitPython.

Lowering the version number is not a good solution, since it will reintroduce the memory problem that caused a lot of trouble a few weeks ago, which was solved by switching to Scipy and using a newer Python version... However, since Python 3.8 has been deprecated for quite some time, and since all current operating systems include at least Python 3.9 and even Python 3.10 or 3.11, I decided to drop support for older versions in favor of supporting all users who had memory problems...

The previous Debian Bulleyes or the newer Debian Bookworm both contain a sufficient version. Debian Buster does indeed lack the correct Python version, so if you are using it, you can always upgrade your OS to something newer or manually install a newer Python version.

Frix-x, 2024-02-01:
However, there is one last point: I have since changed a lot of things in the way S&T works to further improve memory usage and graph generation speed. Of course, this memory problem could also be solved today...

I want to add unit tests to be able to easily test the package against multiple Python versions, and it could be a good opportunity to test which version is compatible and which has this memory problem. This could help to lower the requirements version again, as you've done, but for now I want to keep it since it solved a problem that dozens of users were experiencing and I don't want to introduce a regression...

closed 2024-02-19 by Frix-x