Frix-x/klippain-shaketune #58: Add recommendation to results

pixel-shock, 2024-01-31:
### Describe the feature or hardware support you'd like

Hi,

first of all: GREAT Job!!! Love it!

I think it would be super nice (if possible) to add some kind of recommendation like "you should tighten your B belt" or "you should loosen your B Belt" 

Is that possible based on the date you get from the accelerometer?

Cheers & THX
Tino

Frix-x, 2024-05-19:
Hello,

I left this a bit on the side to work on the main features first. But now that I'm mostly done, I'm taking a closer look at the feature request.

So this is pretty easy to do, but I have some concerns about giving a direct recommendation to tighten A or B belt: since the tension is measured in relative terms, one belt to the other, it's pretty hard to know if it's really B that needs to be tightened, or the opposite A that needs to be loosened. The problem with this is that if there are other problems on the belt path, or if the base tension is already pretty off, you could end up with way too much tension, which could create VFAs, or worse, break printer parts or bend the motor shafts. So I prefer to let the user decide what to do with the information provided.

By the way, to remember which belt is on which side, think of a sheep: BBBAAAAAAA -> B is left, A is right!
I could add a note about this in the documentation and maybe on the graph :)

dezenten, 2024-06-16:
This would be very helpful!
I learned so much about my printer just from using this awesome software and now even more since axes_map_calibration and create_vibrations_profile works.
Thank you Frix-x <3

closed 2024-11-24 by Frix-x