Frix-x/klippain-shaketune #59: axes-map wrong

SnakeKVC, 2024-01-31:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-5-g9fa07a12 

### Describe the bug and expected behavior

axes-map wrong

### Additional information and klippy.log

How can I force the use of axes-map from printer.cfg?

SnakeKVC, 2024-01-31:
My AXES_MAP is -X,Z,Y
AXES_MAP_CALIBRATION show X,Y,Z. It is not right.

Frix-x, 2024-02-01:
As you will see in the console when you use it and in the documentation, this command is very experimental. But that's not a problem because it's just an informative command that does nothing except log some "advice" to the console to help you set up your accelerometer.

If you want to use the info, you'll have to add it to your config manually, as described here: https://www.klipper3d.org/Config_Reference.html#adxl345
But without it, it will not be used anywhere and will just stay as a message in the Klipper console.

On my side, I started working on this feature to try to improve the results and make it more accurate.

closed 2024-02-01 by Frix-x