Frix-x/klippain-shaketune #60: No module named 'pip'

johnrcrc, 2024-01-31:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

i didn't install yet

### Describe the bug and expected behavior

Ihave python version 3.12 and the latest pip update but i get this error:

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

python3-venv is already installed
libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

[SETUP] Virtual environment already exists. Continuing...
[SETUP] Installing/Updating K-Shake&Tune dependencies...
Traceback (most recent call last):
  File "/home/pi/klippain_shaketune-env/bin/pip", line 6, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'

is there any solution?

agravelot, 2024-02-11:
Removing venv folder did the trick fore me:

```bash
$ rm -fr /home/<user>/klippain_shaketune-env
```

Frix-x, 2024-02-12:
Hello,
Yes this happens sometimes when there is some network instability and that all the packages were not properly installed to the venv. Deleting it and restarting the install script should make it work :)

closed 2024-02-12 by Frix-x