Frix-x/klippain-shaketune #63: Graphs not created on Ender5Pro (and working nicely on Ender3v2)

Henk72, 2024-02-12:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-7-gb42e377a

### Describe the bug and expected behavior

When running vibration analysis or shaper, the printer is measuring correctly, but the scripts that make a graph .png file fail with attached error. This happens on my Ender5Pro. On the Ender3v2 all is working as expected.

Could this be related to the fact that the Ender5Pro homes against the max values of the axis (corner right/back) instead of the the Ender3v2 which homes in corner front left (0,0).

### Additional information and klippy.log

main()
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 288, in main
create_vibrations_graph(axis_name=options.axis_name, accel=options.accel_used, chip_name=options.chip_name, keep_csv=options.keep_csv)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 165, in create_vibrations_graph
fig = vibrations_calibration(lognames, KLIPPER_FOLDER, axis_name, accel)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_vibrations.py", line 382, in vibrations_calibration
motor_fr, motor_zeta, motor_max_power_index = compute_mechanical_parameters(motor_vibration_power, freqs)
File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/common_func.py", line 81, in compute_mechanical_parameters
idx_below = np.where(psd[:max_power_index] <= half_power)[0][-1]
IndexError: index -1 is out of bounds for axis 0 with size 0

Henk72, 2024-02-12:
Forgot to mention that the issue is only for the z-axis. The y-axis worked fine.  I tried the axis test script and it gave as answer x, y, -z

wight554, 2024-02-28:
Got exactly same error on ender 3 with K-Shaketune 2.6.0. Might be some cache issue, it was working fine until it randomly stopped working

JenPet3D, 2024-03-01:
Getting also this error message.

Frix-x, 2024-03-04:
I'm not sure about the origin of this error message since it should be an impossible case... Can you share the CSV file that produce this error?

hemasi3d, 2024-03-05:
Hi, 

I have the same problem,

printer.cfg
[stepper_x]
...
position_endstop:260
...
[stepper_y]
...
position_endstop: 150
...

Klippain-version: v2.6.0-2-g312a9c9f
Klipper-version: v0.12.0-115-gb98375b3

[vibr_20240303_175423-sp20_00n1.csv](https://github.com/Frix-x/klippain-shaketune/files/14489989/vibr_20240303_175423-sp20_00n1.csv)

hrvat94, 2024-03-09:
Have a similar Problem: 

Traceback (most recent call last):
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in <module>

    main()
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main

    axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph

    fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration

    performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper

    shaper, all_shapers = helper.find_best_shaper(
TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'

Command {shaketune} finished

Henk72, 2024-03-09:
I re-tested the issue today with the newest version and the issue seems to be solved for me.

Henk72, 2024-03-09:
> I'm not sure about the origin of this error message since it should be an impossible case... Can you share the CSV file that produce this error?

I retested today with the latest version. I got next error again

"motor_fr, motor_zeta, motor_max_power_index = compute_mechanical_parameters(motor_vibration_power, freqs)
File "/home/pi/Ender5Pro_data/config/K-ShakeTuneJos/scripts/common_func.py", line 81, in compute_mechanical_parameters"

"idx_below = np.where(psd[:max_power_index] <= half_power)[0][-1]
IndexError: index -1 is out of bounds for axis 0 with size 0"

I have attached the most recent .csv file from all the generated ones.

[vibr_20240309_173411-sp200_00n1.csv](https://github.com/Frix-x/klippain-shaketune/files/14547104/vibr_20240309_173411-sp200_00n1.csv)

Frix-x, 2024-03-11:
@hrvat94 your issue is different and it's because you have updated Shake&Tune to v2.6 and will need to also update Klipper to a more recent version. This will fix your issue. I hid your comment to keep the topic on the current issue.

Henk72, 2024-03-11:
Dear Frix-x,

This is my full set of CSV files zipped. The problem happens on the Y-axis of the Ender5Pro. No issues on my Ender3v2.
[vibrations.zip](https://github.com/Frix-x/klippain-shaketune/files/14555905/vibrations.zip)

Frix-x, 2024-04-30:
Hello, this was fixed in v3 and should not happen anymore :)