Frix-x/klippain-shaketune #65: Subprocess exited with error

bewst, 2024-02-14:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

N/A

### Describe the bug and expected behavior

This is during a new install - subprocess-exited-with-error.

pip subprocess to install build dependencies did not run successfully. Initially similar to this https://github.com/Frix-x/klippain-shaketune/issues/30 so followed those steps. 

MCU is a 2040 (Banana Pi M2 Zero)  
Version: v0.12.0-65-g96ab9069-dirty
OS: Armbian 21.08.1 Bullseye 

Of note during the install:

  FAILED: numpy/core/libloops_comparison.dispatch.h_NEON.a.p/meson-generated_loops_comparison.dispatch.c.o
            cc -Inumpy/core/libloops_comparison.dispatch.h_NEON.a.p -Inumpy/core -I../numpy/core -Inumpy/core/include -I../numpy/core/include -I../numpy/core/src/common -I../numpy/core/src/multiarray -I../numpy/core/src/npymath -I../numpy/core/src/umath -I/usr/include/python3.9 -I/usr/include/arm-linux-gnueabihf/python3.9 -I/tmp/pip-install-vzvep7e0/numpy_3ebe65d4b8454c02b5383b87282a2384/.mesonpy-a_sg6v9t/meson_cpu -fdiagnostics-color=always -DNDEBUG -Wall -Winvalid-pch -std=c99 -O3 -fno-strict-aliasing -fPIC -DNPY_INTERNAL_BUILD -DHAVE_NPY_CONFIG_H -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -O3 -DNPY_HAVE_NEON -mfpu=neon -DNPY_MTARGETS_CURRENT=NEON -MD -MQ numpy/core/libloops_comparison.dispatch.h_NEON.a.p/meson-generated_loops_comparison.dispatch.c.o -MF numpy/core/libloops_comparison.dispatch.h_NEON.a.p/meson-generated_loops_comparison.dispatch.c.o.d -o numpy/core/libloops_comparison.dispatch.h_NEON.a.p/meson-generated_loops_comparison.dispatch.c.o -c numpy/core/libloops_comparison.dispatch.h_NEON.a.p/loops_comparison.dispatch.c
            cc: fatal error: Killed signal terminated program cc1
            compilation terminated.

[install_info.txt](https://github.com/Frix-x/klippain-shaketune/files/14274421/install_info.txt)

### Additional information and klippy.log

Text file attached

Frix-x, 2024-02-14:
Hi,

If you have followed the info, you may want to recreate the venv. For this, just delete the `~/klippain_shaketune-env` folder and then start again the install script.

Let me know if it solved the problem

bewst, 2024-02-15:
Thanks Frix-x, 

Tried re-creating env by deleting and ran the install again. I got further along but still failed. It may just be my lack of compute on this banana pi. Tried creating a swap shell and running the installer there but no luck either, CPU and MEM look to be maxed causing the MCU to crash. 
[install_2.txt](https://github.com/Frix-x/klippain-shaketune/files/14293187/install_2.txt)

Frix-x, 2024-02-19:
It indeed look like compilation issues due to high RAM usage... I was able to make it work on a Pi zero 2W but maybe some wheels were available for this platform that avoid compilation and that is not the case for the banana pi... I'm sorry but I don't know how to help you on this...

I'll let the issue open a couple of days if you find out a solution

Life0fBrian, 2024-03-14:
You can create a bigger swap file on your SD card to increase the virtual RAM.
This will take ages to install/compile numpy but I got it done on an Orange Pi...

closed 2024-03-24 by Frix-x