Frix-x/klippain-shaketune #66: Multi-instance installation support

NNGB-kor, 2024-02-15:
### Describe the feature or hardware support you'd like

Cannot install in a multi-instance configuration.
[ERROR] Klipper service not found, please install Klipper first!
In my case, printer1_data printer2_data.... I have many printers in the following format
The install script only supports single instant, so it seems to need to be modified in some way.

Frix-x, 2024-02-18:
Yes, S&T installer doesn't support multi-klipper instances since it's not a conventional way to install Klipper...

I do plan to work on this in the future, but not today. In the meantime, if you want to still use S&T, you can manually place the scripts in the correct folders and copy/paste the macros in you config. It should work if you fix the paths :)

Frix-x, 2024-02-18:
I'll close this issue since it's a duplicate of #12

closed 2024-02-18 by Frix-x

grilparto, 2024-05-29:
Not my intention to re-open this issue.  Just wanted to add some extra info for anyone looking for a workaround.

To make install easier, you'll want to just grab the install.sh and modify it a little.

`wget https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh`

edit the first two variables of the install.sh to reflect the path to your machine's config folder.

`USER_CONFIG_PATH="${HOME}/ender5plus_data/config"`
`MOONRAKER_CONFIG="${HOME}/ender5plus_data/config/moonraker.conf"`

In this case, my config folder is in /home/pi/ender5plus_data (hence ${HOME}/ender5plus_data/config/)

Next you'll need to make it executable. `sudo chmod +x install.sh`

Finally, run the install.sh.  `./install.sh`

At this point the install script should execute successfully.  It will fail to restart the klipper service (probably) but you're likely going to be restarting klipper anyway after you edit your printer.cfg file to include the macros.  That brings us to one final step.  We need to edit **shaketune_cmd.cfg** or else the shaketune command will fail to execute automatically after running any of the tests.  

You should be able to access the cfg file from Mainsail for Fluidd.  In the config/K-ShakeTune folder, open shaketune_cmd.cfg and make the following change to the "command:" line to reflect the path to your machine's config folder.

`command: ~/ender5plus_data/config/K-ShakeTune/shaketune.sh`

Again, mine is ender5plus_data, insert your own config folder's path.

That should allow everything to run properly with **ONE FINAL EXCEPTION**!  **Where are the resulting files?!**  Well.  I don't rightly know what to change to workaround this one, but the shaketune script will just create the files in the default location you'd expect for one instance of klipper.  `/printer_data/config/K-ShakeTune_results`.  So, fire up WinSCP or whatever and go find them there.

Hope that helps folks with multiple instances!  I guess it really only installs for just the one instance out of many, but I guess I don't see why one couldn't just execute the install.sh again for each instance--each time manually changing the config folder location using the steps above.

eaman, 2024-12-29:
> edit the first two variables of the install.sh to reflect the path to your machine's config folder.

On my system it still fails to find the correct klippy service, in the install script there's a check for:

`systemctl list-units --full -all -t service --no-legend | grep -F 'klipper.service'`

It has to be changed according to the qualified name of your klippy printer instance, in my case "klipper-Ender3.service" as:

`    if [ "$(sudo systemctl list-units --full -all -t service --no-legend | grep -F 'klipper-Ender3.service')" ]; then
        printf "[PRE-CHECK] Klipper service found! Continuing...\n\n"`

Or erase the check.

sbhanji, 2025-05-14:
Unfortunately none of these worked in my system, nothing was created in the config directory. When trying to troubleshoot, I noticed that some of the *.py files have hard-coded values to {$HOME}/printer_data/...