Frix-x/klippain-shaketune #67: No more result images

insane78, 2024-02-17:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.5.0-7-gb42e377a

### Describe the bug and expected behavior

[image]

therick0996, 2024-02-18:
i'm also not getting graphs with input shaper. i get graphs for belt shaper, however. 

brand new install of Klipper and shaketune

> 19:29:01 
// Command {shaketune} finished
19:29:01 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
// df = math.sqrt(1. - damping_ratio**2)
// ~~~~~~~~~~~~~^^~
// TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
19:29:01 
// shaper = self.background_process_exec(self.fit_shaper, (
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
// return method(*args)
// ^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
// shaper = shaper_cfg.init_func(test_freq, damping_ratio)
19:29:01 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper
19:29:01 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 51, in calibrate_shaper
// shaper, all_shapers = helper.find_best_shaper(calibration_data, max_smoothing, print_with_c_locale)
19:29:01 
// performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing)
19:29:01 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 212, in shaper_calibration
19:29:01 
// fig = shaper_calibration([new_file], KLIPPER_FOLDER)
19:29:01 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
19:29:01 
// axis = create_shaper_graph(keep_csv=options.keep_csv)
19:29:01 
// main()
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 285, in main
19:29:01 
// Traceback (most recent call last):
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 299, in
19:28:56 
// Running Command {shaketune}...:
19:28:56 
echo: This may take some time (1-3min)
19:28:56 
echo: Y axis frequency profile generation...
19:28:56 
// Writing raw accelerometer data to /tmp/raw_data_y_y.csv file

----------------------------------------------
> 19:26:47 
// Command {shaketune} finished
19:26:46 
// shaper = shaper_cfg.init_func(test_freq, damping_ratio)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
// df = math.sqrt(1. - damping_ratio**2)
// ~~~~~~~~~~~~~^^~
// TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
19:26:46 
// return method(*args)
// ^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
19:26:46 
// shaper = self.background_process_exec(self.fit_shaper, (
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
19:26:46 
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper
19:26:46 
// axis = create_shaper_graph(keep_csv=options.keep_csv)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
// fig = shaper_calibration([new_file], KLIPPER_FOLDER)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 212, in shaper_calibration
// performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 51, in calibrate_shaper
// shaper, all_shapers = helper.find_best_shaper(calibration_data, max_smoothing, print_with_c_locale)
19:26:46 
// main()
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 285, in main
19:26:46 
// Traceback (most recent call last):
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 299, in
19:26:42 
// Running Command {shaketune}...:
19:26:42 
echo: This may take some time (1-3min)
19:26:42 
echo: X axis frequency profile generation...
19:26:42 
// Writing raw accelerometer data to /tmp/raw_data_x_x.csv file

Heartlander81, 2024-02-18:
i'm also not getting graphs with input shaper. i get graphs for belt shaper

beavis69, 2024-02-18:
Hello,

same issue here.

```
Traceback (most recent call last):
  File "/home/linaro/printer_data/config/scripts/is_workflow/is_workflow.py", line 231, in <module>

    main()
  File "/home/linaro/printer_data/config/scripts/is_workflow/is_workflow.py", line 217, in main
    fig, png_filename = get_shaper_graph()
  File "/home/linaro/printer_data/config/scripts/is_workflow/is_workflow.py", line 116, in get_shaper_graph

    fig = shaper_calibration([new_file], KLIPPER_FOLDER)
  File "/home/linaro/klippain_config/scripts/is_workflow/graph_shaper.py", line 335, in shaper_calibration
    performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper_with_damping(datas, max_smoothing)
  File "/home/linaro/klippain_config/scripts/is_workflow/graph_shaper.py", line 82, in calibrate_shaper_with_damping
    shaper, all_shapers = helper.find_best_shaper(calibration_data, max_smoothing, print)
  File "/home/linaro/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper

    shaper = self.background_process_exec(self.fit_shaper, (
  File "/home/linaro/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
    return method(*args)
  File "/home/linaro/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
    shaper = shaper_cfg.init_func(test_freq, damping_ratio)
  File "/home/linaro/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper

    df = math.sqrt(1. - damping_ratio**2)
TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
```
Seems to be related to klipper version v0.12.0-113
If i rollback to klipper  v0.12.0-111 it's working again

insane78, 2024-02-18:
> Hello,
>
> same issue here.
>
> ```
> Traceback (most recent call last):
>   File "/home/linaro/printer_data/config/scripts/is_workflow/is_workflow.py", line 231, in <module>
>
>     main()
>   File [...]

Yes mate, same klipper version here

UnifiBg, 2024-02-18:
Just saying same issue.

Frix-x, 2024-02-18:
Hello everyone, it seems that this issue is related to an update of Klipper two days ago. I'll update my machine and try to reproduce and fix it.

insane78, 2024-02-18:
> Hello everyone, it seems that this issue is related to an update of Klipper two days ago. I'll update my machine and try to reproduce and fix it.

Thanks <3

therick0996, 2024-02-19:
i'm guessing [this commit](https://github.com/Klipper3d/klipper/commit/72b301a2859c3f7ed26d802dd52fc495eef6c353) has something to do with it

Frix-x, 2024-02-19:
> i'm guessing [this commit](https://github.com/Klipper3d/klipper/commit/72b301a2859c3f7ed26d802dd52fc495eef6c353) has something to do with it

Yes exactly. I just released a new version of S&T that should fix the issue. Please let me know if it's ok for you :)

This new version also has pretty good new features based on the Klipper change that will allow to have results much more precise and representative of what will happen in your machine (see [release notes](https://github.com/Frix-x/klippain-shaketune/releases/tag/v2.6.0))

Frix-x, 2024-02-19:
I got confirmation that it fixed the problem. So I'll close this issue now :)