Frix-x/klippain-shaketune #68: Axes shaper calibration graph plot fails with: "TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'"

pvpacheco, 2024-02-20:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.0-1-gf4e700a1

### Describe the bug and expected behavior

When running axes shaper calibration the raw accelerometer csv files are generated correctly, but the graph plot script fails with attached error. 

### Additional information and klippy.log

```
File "/home/ppacheco/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper
shaper = self.background_process_exec(self.fit_shaper, (
File "/home/ppacheco/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
return method(*args)
File "/home/ppacheco/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
shaper = shaper_cfg.init_func(test_freq, damping_ratio)
File "/home/ppacheco/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
df = math.sqrt(1. - damping_ratio**2)
TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
```

Frix-x, 2024-02-20:
This error was solved in S&T v2.6 and was due to a breaking change on Klipper side. Are you sure you are using the latest version?

josoli1968, 2024-02-20:
I have same error. Klipper Version is v0.12.0-113-g28f06a10.
This is my error:

return method(*args)
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
shaper = shaper_cfg.init_func(test_freq, damping_ratio)
File "/home/pi/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
df = math.sqrt(1. - damping_ratio**2)
TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
16:12:28
main()
File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 217, in main
fig, png_filename = get_shaper_graph()
File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 116, in get_shaper_graph
fig = shaper_calibration([new_file], KLIPPER_FOLDER)
File "/home/pi/klippain_config/scripts/is_workflow/graph_shaper.py", line 335, in shaper_calibration
performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper_with_damping(datas, max_smoothing)
File "/home/pi/klippain_config/scripts/is_workflow/graph_shaper.py", line 82, in calibrate_shaper_with_damping
shaper, all_shapers = helper.find_best_shaper(calibration_data, max_smoothing, print)
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper
shaper = self.background_process_exec(self.fit_shaper, (
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec

Frix-x, 2024-02-20:
> I have same error. Klipper Version is v0.12.0-113-g28f06a10.

On your case it look like you are using the full Klippain config that doesn't have the latest S&T integrated... yet. It's done on the develop branch and I do plan on releasing it very soon (this week).

M3HNGRY, 2024-02-21:
> > I have same error. Klipper Version is v0.12.0-113-g28f06a10.
> 
> On your case it look like you are using the full Klippain config that doesn't have the latest S&T integrated... yet. It's done on the develop branch and I do plan on releasing it very soon (this week).

Ahh ok this will fix my problem then :D I will wait.

Frix-x, 2024-02-21:
Ok so now I understand better, and it's "normal"...

> Is there anyway to change klippain to look for klippain-shaketune path when executing those scripts?

Not really without tempering with the default Klippain files and making the repo "dirty". But please wait a day or two if possible, I'm really near the final release of the new version (In fact, I just need to find time to write the changelog and do one or two tests and it will be ok)

closed 2024-02-21 by Frix-x