Frix-x/klippain-shaketune #69: TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'

TItan371, 2024-02-20:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.0-1-gf4e700a1

### Describe the bug and expected behavior

no PNG output

### Additional information and klippy.log

11:47:29 AM 
// Testing frequency 97 Hz
11:47:30 AM 
// Testing frequency 98 Hz
11:47:31 AM 
// Testing frequency 99 Hz
11:47:32 AM 
// Testing frequency 100 Hz
11:47:33 AM 
// Testing frequency 101 Hz
11:47:34 AM 
// Testing frequency 102 Hz
11:47:35 AM 
// Testing frequency 103 Hz
11:47:36 AM 
// Testing frequency 104 Hz
11:47:37 AM 
// Testing frequency 105 Hz
11:47:38 AM 
// Testing frequency 106 Hz
11:47:39 AM 
// Testing frequency 107 Hz
11:47:40 AM 
// Testing frequency 108 Hz
11:47:41 AM 
// Testing frequency 109 Hz
11:47:42 AM 
// Testing frequency 110 Hz
11:47:43 AM 
// Testing frequency 111 Hz
11:47:44 AM 
// Testing frequency 112 Hz
11:47:45 AM 
// Testing frequency 113 Hz
11:47:46 AM 
// Testing frequency 114 Hz
11:47:47 AM 
// Testing frequency 115 Hz
11:47:48 AM 
// Testing frequency 116 Hz
11:47:49 AM 
// Testing frequency 117 Hz
11:47:50 AM 
// Testing frequency 118 Hz
11:47:51 AM 
// Testing frequency 119 Hz
11:47:52 AM 
// Testing frequency 120 Hz
11:47:53 AM 
// Testing frequency 121 Hz
11:47:54 AM 
// Testing frequency 122 Hz
11:47:55 AM 
// Testing frequency 123 Hz
11:47:56 AM 
// Testing frequency 124 Hz
11:47:57 AM 
// Testing frequency 125 Hz
11:47:58 AM 
// Testing frequency 126 Hz
11:47:59 AM 
// Testing frequency 127 Hz
11:48:00 AM 
// Testing frequency 128 Hz
11:48:01 AM 
// Testing frequency 129 Hz
11:48:02 AM 
// Testing frequency 130 Hz
11:48:03 AM 
// Testing frequency 131 Hz
11:48:04 AM 
// Testing frequency 132 Hz
11:48:05 AM 
// Testing frequency 133 Hz
11:48:06 AM 
// Re-enabled [input_shaper]
11:48:08 AM 
// Writing raw accelerometer data to /tmp/raw_data_y_y.csv file
11:48:08 AM 
echo: Y axis frequency profile generation...
11:48:08 AM 
echo: This may take some time (1-3min)
11:48:08 AM 
// Running Command {shaketune}...:
11:48:12 AM 
// Traceback (most recent call last):
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in
11:48:12 AM 
// main()
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main
11:48:12 AM 
// axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
11:48:12 AM 
// fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration
11:48:12 AM 
// performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
// File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper
11:48:13 AM 
// shaper, all_shapers = helper.find_best_shaper(
// TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'
11:48:13 AM 
// Command {shaketune} finished
11:48:13 AM 
// Running Command {shaketune}...:
11:48:15 AM 
// Cleaning output folder to keep only the last 3 results...
11:48:16 AM 
// Command {shaketune} finished

spigotdesign, 2024-02-20:
Getting this same error.

`Traceback (most recent call last):
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in <module>

    main()
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main

    axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
    fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration

    performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
  File "/home/pi/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper
    shaper, all_shapers = helper.find_best_shaper(

TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'`

Frix-x, 2024-02-20:
**Please update Klipper to a version after Feb 17, 2024.**

Indeed, Klipper introduced some changes in the way resonance testing work that broke S&T. I had to make modifications to S&T in order to support this, but this also means that you need to update Klipper.

D1m0k, 2024-02-20:
i confirm, klipper update fix this bug

Frix-x, 2024-02-20:
I've added some notes in the install doc and release notes to specify this :)

closed 2024-02-20 by Frix-x