Frix-x/klippain-shaketune #70: Error loading template 'gcode_macro AXES_SHAPER_CALIBRATION

chopdi, 2024-02-20:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

Newest

### Describe the bug and expected behavior

Klipper reports: ERROR

Error loading template 'gcode_macro AXES_SHAPER_CALIBRATION:gcode': jinja2.exceptions.TemplateAssertionError: no test named 'None'

Nukle0n, 2024-02-20:
Also getting this bug when initializing klipper

`Error loading template 'gcode_macro AXES_SHAPER_CALIBRATION:gcode': jinja2.exceptions.TemplateAssertionError: no test named 'None'
`

MickCZ, 2024-02-20:
I have the same problem after updating to v2.6.0-0-g80c8da62

Frix-x, 2024-02-20:
Please update to S&T >v2.6.0-1 as this bug is already solved and was a typo in the macro. If you don't have the correct version in Moonraker update manager, please validate that the moonraker.conf section is ok (you can find the correct one here: https://github.com/Frix-x/klippain-shaketune/blob/main/moonraker.conf)

Nukle0n, 2024-02-20:
> Please update to S&T >v2.6.0-1 as this bug is already solved and was a typo in the macro. If you don't have the correct version in Moonraker update manager, please validate that the moonraker.conf [...]

Thanks for the quick fix

closed 2024-02-21 by Frix-x