Frix-x/klippain-shaketune #71: After Running AXES_MAP_CALIBRATION i do not get the graphs

ghostv714, 2024-02-20:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.0-2-g312a9c9f 

### Describe the bug and expected behavior

when i call AXES_MAP_CALIBRATION. the test begins to completion i get no errors. then when it comes time to get the results and the pngs all i get is the cvs.
this is what i get when the printer is done moving:
` 10:13:24  // Command {shaketune} finished
10:13:23  // performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
// File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper
// shaper, all_shapers = helper.find_best_shaper(
// TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'
10:13:23  // axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
// File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
// fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
// File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration
10:13:23  // main()
// File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main
10:13:23  // Traceback (most recent call last):
// File "/home/biqu/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in
10:13:14  // Running Command {shaketune}...:
10:13:14  echo: This may take some time (1-3min)
10:13:14  echo: Y axis frequency profile generation...
10:13:14  // Writing raw accelerometer data to /tmp/raw_data_y_y.csv file
10:13:11  // Re-enabled [input_shaper]`

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/14349934/klippy.log)

M3HNGRY, 2024-02-21:
I'm getting same errors as well..

https://cdn.discordapp.com/attachments/1096701708643614820/1209707525155463168/image.png?ex=65e7e6ed&is=65d571ed&hm=17bb824f3f3e338a6489e1f69bbb7be8b853c9bddd53cf9ab0c41b66a54b4ffe&

https://cdn.discordapp.com/attachments/1096701708643614820/1209707632185712670/image.png?ex=65e7e706&is=65d57206&hm=9af3e6b93ee8d267c51e507c4d20478746f88ac6d836f6abd8a07b88d0df13ed&

ghostv714, 2024-02-21:
You know I didn't think of it but I am running klipper on a BTT CB1. Would that have anything to do with my errors

skyaee, 2024-03-01:
[image]
I have the latest version and have the same problem. I cannot display png. Is it because I switched to danger-klipper?

closed 2024-03-04 by Frix-x

dougshell, 2024-03-19:
I have the same behaviuor running the latest klipper

Phil1988, 2024-04-04:
So did I get it right, that S&T is (in the latest version) not any more compatible to the latest klipper?

I think the last working version was v2.6.0 is this right?
Im asking because I think I may have this problem when perfoming "AXES_SHAPER_CALIBRATION":
[image]