Frix-x/klippain-shaketune #73: Error loading template 'gcode_macro AXES_SHAPER_CALIBRATION

axg20202, 2024-02-23:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

2.6.0

### Describe the bug and expected behavior

I just updated to latest version of klipper and shaketune, and now klipper will not start. I get the following error:
Error loading template 'gcode_macro AXES_SHAPER_CALIBRATION:gcode': jinja2.exceptions.TemplateAssertionError: no test named 'None'

Any ideas?

locki-cz, 2024-02-23:
this is rollback to previous version:

```
cd klippain_shaketune
git reset --hard HEAD~1
```

Frix-x, 2024-02-25:
This is a duplicate of #70 and #72 . This error was a typo fixed a couple of days ago. Please update to at least S&T v2.6.0-1 :)

If you don't see it, update your moonraker.conf with the default one from this repo here: https://github.com/Frix-x/klippain-shaketune/blob/main/moonraker.conf

closed 2024-03-04 by Frix-x