Frix-x/klippain-shaketune #75: ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE.

lunarconcepts, 2024-03-09:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.0

### Describe the bug and expected behavior

Hello, Thanks for your work on Shaketune . I am new to github format so I apologize if this isn't how to handle these issue. I am gettin the listed error when trying to add shaketune via putty. Maybe I am doing something wrong but seems you made it as easy as possible? 
[image]

V1EngineeringInc, 2024-03-09:
I had that happen yesterday, it just installed fine today. Might want to give it another try.

Great software!

lunarconcepts, 2024-03-09:
> I had that happen yesterday, it just installed fine today. Might want to give it another try.
> 
> Great software!

Thanks, I am up and running now!

V1EngineeringInc, 2024-03-09:
Might want to close the issue, that helps out the dev.

Frix-x, 2024-03-11:
This usually happens when the network connection is a bit unstable. Usually clearing the pip cache and trying again solve this issue :)

closed 2024-03-11 by Frix-x