Frix-x/klippain-shaketune #76: unexpected keyword argument

AllMx, 2024-03-10:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

2.5.0

### Describe the bug and expected behavior

ShaperCalibrate.find_best_shaper() got an unexpected keyword argument 'shapers'

### Additional information and klippy.log

Hello, I installed version 2.5.0 and when I run AXES_SHAPER_CALIBRATION I get this problem:

 14:26:07  $ AXES_SHAPER_CALIBRATION AXIS=Y FREQ_END=10
14:26:10  // Disabled [input_shaper] for resonance testing
14:26:10  // Testing frequency 5 Hz
14:26:10  // Testing frequency 6 Hz
14:26:10  // Testing frequency 7 Hz
14:26:10  // Testing frequency 8 Hz
14:26:10  // Testing frequency 9 Hz
14:26:13  // Testing frequency 10 Hz
14:26:13  // Re-enabled [input_shaper]
14:26:15  // Writing raw accelerometer data to /tmp/raw_data_y_y.csv file
14:26:15  echo: Y axis frequency profile generation...
14:26:15  echo: This may take some time (1-3min)
14:26:15  // Running Command {shaketune}...:
14:26:21  // Traceback (most recent call last):
// File "/home/mks/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in
14:26:21  // main()
// File "/home/mks/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main
14:26:21  // axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/mks/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
// fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/mks/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration
// performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
// ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
// File "/home/mks/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper
// shaper, all_shapers = helper.find_best_shaper(
// ^^^^^^^^^^^^^^^^^^^^^^^^
// TypeError: ShaperCalibrate.find_best_shaper() got an unexpected keyword argument 'shapers'
14:26:22  // Command {shaketune} finished
14:26:22  // Running Command {shaketune}...:
14:26:27  // Cleaning output folder to keep only the last 3 results...

I can't figure out what's wrong. Distribution Debian Buster, Python 3.12.2

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/14549973/klippy.log)

Frix-x, 2024-03-11:
This is because you are using S&T v2.5 with a newer Klipper version. Please update to S&T v2.6+ to make it work correctly or downgrade Klipper from before the 17 Feb

AllMx, 2024-03-11:
I use elegoo klipper (printer neptune 4) and at first I installed the latest version of klippain-shaketune. Had the same error, so I decided to downgrade to v2.5

Frix-x, 2024-03-14:
If you have the same error on the latest, then it means that you need to update Klipper to a more recent version

berlin4apk, 2024-03-15:
An update of Klipper on the neptune 4 is not really possible, as it has some eleggo customized things in it.

closed 2024-04-30 by Frix-x