Frix-x/klippain-shaketune #78: SET_VELOCITY_LIMIT is deprecated

XEGARE, 2024-03-14:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.0-3-g536c3c0e

### Describe the bug and expected behavior

Hello. In the new klipper v0.12.0-124-g09a78c31 `SET_VELOCITY_LIMIT` is deprecated. `max_accel_to_decel` too

meteyou, 2024-03-14:
`SET_VELOCITY_LIMIT` is not deprecated, only the `max_accel_to_decel` attribute. The new attribute is called `minimum_cruise_ratio`.

Klipper PR: https://github.com/Klipper3d/klipper/pull/6418

and here the info in the `Config_changes` page from Klipper (https://www.klipper3d.org/Config_Changes.html#changes):
```
20240313: The max_accel_to_decel parameter in the [printer] config section has been deprecated. The ACCEL_TO_DECEL parameter of the SET_VELOCITY_LIMIT command has been deprecated. The printer.toolhead.max_accel_to_decel status has been removed. Use the [minimum_cruise_ratio parameter](https://www.klipper3d.org/Config_Reference.html#printer) instead. The deprecated features will be removed in the near future, and using them in the interim may result in subtly different behavior.
```

meteyou, 2024-03-14:
I found already a PR for this: https://github.com/Frix-x/klippain-shaketune/pull/53

so it looks like it will be fixed very soon...

Frix-x, 2024-03-15:
Yes, I was not available until now. I merge it :)