Frix-x/klippain-shaketune #80: BELTS_SHAPER_CALIBRATION: Error running command {shaketune}

mokshmridul, 2024-03-19:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.1-0-g82b91c1b

### Describe the bug and expected behavior

After running `BELTS_SHAPER_CALIBRATION`, the printer does the test for both axis, but ends with the `Error running command {shaketune}`  without generating graphs.

Have tried rebooting the printer, removing and rebooting and then reinstalling. All to no avail. 

### Additional information and klippy.log

[image]

Frix-x, 2024-03-19:
Can you share the klippy.log when the error happens to allow me investigating the issue?

Thanks

mokshmridul, 2024-03-20:
Apologies for not attaching the klippy.log earlier. Here it is.
[klippy (1).log](https://github.com/Frix-x/klippain-shaketune/files/14662112/klippy.1.log)

Frix-x, 2024-03-20:
Ok, you have something strange happening... Probably because you transfered everything from a USB key or something?

But the error is: `PermissionError: [Errno 13] Permission denied: '/home/moksh/printer_data/config/K-ShakeTune/scripts/shaketune.sh'`

So to fix it, open an SSH prompt and type: `chmod +x /home/moksh/printer_data/config/K-ShakeTune/scripts/shaketune.sh`

mokshmridul, 2024-03-20:
it was working fine after i transferred over, but then it updated and i got stuck. weird.

This solved the issue though. Thank you :)

closed 2024-03-20 by mokshmridul