Frix-x/klippain-shaketune #81: Running Command {shaketune}...: Illegal instruction

Snuf1337, 2024-03-21:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.1-1-gbf7a98d9

### Describe the bug and expected behavior

When I use the macro, the following error always occurs:

/home/sonic/printer_data/config/K-ShakeTune/scripts/shaketune.sh: line 4: 15380 Illegal instruction python ~/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py "$@"

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/14711767/klippy.log)

ozelentok, 2024-03-24:
There are no native modules in ShakeTune so the error must be originating from one of the python dependencies, possibly numpy or openblas
Checkout [the thread on the Klipper Discourse](https://klipper.discourse.group/t/klipper-crashes-when-running-resonance-measurements/13656) for more details

Snuf1337, 2024-03-24:
I can use the normal Inputshaper script from Klipper and get a graphic created but the script does not work

ozelentok, 2024-03-24:
Try the following commands, one of the imports should fail
```bash
source ~/klippain_shaketune-env/bin/activate
pip list

python -c "import matplotlib"
python -c "import numpy"
python -c "import scipy'
```
Since klipper internal graphing works I believe Klipper is using a different version that is compatible with the cpu

Snuf1337, 2024-03-25:
This is the output:

Linux SonicPad 4.9.191 #3 SMP PREEMPT Sun Jun 18 03:21:17 UTC 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar 24 19:46:10 2024 from 192.168.178.21
sonic@SonicPad:~$ source ~/klippain_shaketune-env/bin/activate
pip list
Package             Version
------------------- -----------
contourpy           1.2.0
cycler              0.12.1
fonttools           4.50.0
gitdb               4.0.11
GitPython           3.1.40
importlib_resources 6.4.0
kiwisolver          1.4.5
matplotlib          3.8.2
numpy               1.26.2
packaging           24.0
pillow              10.2.0
pip                 24.0
pkg_resources       0.0.0
pyparsing           3.1.2
python-dateutil     2.9.0.post0
scipy               1.11.4
setuptools          44.1.1
six                 1.16.0
smmap               5.0.1
zipp                3.18.1
(klippain_shaketune-env) sonic@SonicPad:~$ python -c "import matplotlib"
Illegal instruction
(klippain_shaketune-env) sonic@SonicPad:~$ python -c "import numpy"
(klippain_shaketune-env) sonic@SonicPad:~$ python -c "import scipy'

Frix-x, 2024-04-30:
The way Shake&Tune is started is different now and this issue is now outdated. Can you try again on v3 ?

closed 2024-05-05 by Frix-x