Frix-x/klippain-shaketune #82: S&T will not create the graph

tecno-roy, 2024-03-24:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

2.6.1

### Describe the bug and expected behavior

I get the following error:
shaper = shaper_cfg.init_func(test_freq, damping_ratio)
File "/home/pi/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
df = math.sqrt(1. - damping_ratio**2)
TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/14735925/klippy.log)

Frix-x, 2024-03-24:
It seems that you are using an older version of Klipper and you will need to update it. Indeed, due to some breaking changes in the resonance testing code on the Klipper side, Shake&Tune has been modified to take advantage of this, and thus S&T v2.6+ will only support a Klipper version from Feb 17th 2024. If you are using an older version of Klipper, you must use S&T <=2.5.x

tecno-roy, 2024-03-27:
Sorry for my late answere...
My Klipper version is v0.12.0-132-ge37b007f

Frix-x, 2024-04-11:
I have added a compatibility mode for the next planned release of S&T. So even if you should update Klipper to the latest version to benefit from all the new features, it will not fail anymore and will run correctly even with older Klipper versions.
I'll close the issue when v3.0.0 hit the main branch :)