Frix-x/klippain-shaketune #83: S&T new version not generating Graph after running.

einsteins12, 2024-03-28:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.1-1-gbf7a98d9

### Describe the bug and expected behavior

I upgraded to the newest version and all runs well accept that it does not generate any graphs. 
In the results folder there is only the .CSV file(s)

### Additional information and klippy.log

Klippy File to large to upload.

einsteins12, 2024-03-28:
[image]

The-Original-Reth, 2024-03-28:
This is from einsteins klippy log

Running Command {shaketune}...:
ADXL345 finished 'adxl345' measurements
Stats 341387.5: gcodein=0  mcu: mcu_awake=0.054 mcu_task_avg=0.000014 mcu_task_stddev=0.000010 bytes_write=13237699 bytes_read=3292299 bytes_retransmit=9 bytes_invalid=0 send_seq=264311 receive_seq=264311 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=179999348 can0: mcu_awake=2.413 mcu_task_avg=0.000126 mcu_task_stddev=0.000000 bytes_write=6780226 bytes_read=6013702 bytes_retransmit=0 bytes_invalid=0 send_seq=334859 receive_seq=334859 retransmit_seq=0 srtt=0.002 rttvar=0.002 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998598 adj=63998896  heater_bed: target=115 temp=115.0 pwm=0.288 raspberry_pi: temp=46.7 mcu_temp: temp=38.5 CANBOARD: temp=75.2 chamber: temp=47.9 sysload=0.28 cputime=44230.963 memavail=395944 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=69.7 pwm=0.000
Stats 341388.5: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000013 mcu_task_stddev=0.000010 bytes_write=13237855 bytes_read=3292708 bytes_retransmit=9 bytes_invalid=0 send_seq=264322 receive_seq=264322 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=179999351 can0: mcu_awake=2.413 mcu_task_avg=0.000126 mcu_task_stddev=0.000000 bytes_write=6780836 bytes_read=6014218 bytes_retransmit=0 bytes_invalid=0 send_seq=334897 receive_seq=334897 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998592 adj=63998831  heater_bed: target=115 temp=115.1 pwm=0.288 raspberry_pi: temp=48.2 mcu_temp: temp=38.5 CANBOARD: temp=75.1 chamber: temp=48.0 sysload=0.28 cputime=44231.073 memavail=371160 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=69.8 pwm=0.000
Stats 341389.5: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000013 mcu_task_stddev=0.000010 bytes_write=13238001 bytes_read=3293115 bytes_retransmit=9 bytes_invalid=0 send_seq=264333 receive_seq=264333 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=179999347 can0: mcu_awake=2.413 mcu_task_avg=0.000126 mcu_task_stddev=0.000000 bytes_write=6781458 bytes_read=6014795 bytes_retransmit=0 bytes_invalid=0 send_seq=334943 receive_seq=334943 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998586 adj=63998731  heater_bed: target=115 temp=115.1 pwm=0.288 raspberry_pi: temp=49.2 mcu_temp: temp=38.4 CANBOARD: temp=75.1 chamber: temp=47.9 sysload=0.49 cputime=44231.186 memavail=361668 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=69.8 pwm=0.000
Stats 341390.5: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000013 mcu_task_stddev=0.000010 bytes_write=13238163 bytes_read=3293498 bytes_retransmit=9 bytes_invalid=0 send_seq=264345 receive_seq=264345 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=179999346 can0: mcu_awake=2.413 mcu_task_avg=0.000126 mcu_task_stddev=0.000000 bytes_write=6782110 bytes_read=6015356 bytes_retransmit=0 bytes_invalid=0 send_seq=334990 receive_seq=334990 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998581 adj=63998672  heater_bed: target=115 temp=115.1 pwm=0.249 raspberry_pi: temp=50.1 mcu_temp: temp=38.5 CANBOARD: temp=75.0 chamber: temp=47.9 sysload=0.49 cputime=44231.292 memavail=420512 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=69.9 pwm=0.000
Stats 341391.5: gcodein=0  mcu: mcu_awake=0.035 mcu_task_avg=0.000013 mcu_task_stddev=0.000010 bytes_write=13238309 bytes_read=3293891 bytes_retransmit=9 bytes_invalid=0 send_seq=264356 receive_seq=264356 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=179999339 can0: mcu_awake=2.413 mcu_task_avg=0.000126 mcu_task_stddev=0.000000 bytes_write=6782756 bytes_read=6015906 bytes_retransmit=0 bytes_invalid=0 send_seq=335039 receive_seq=335038 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63998574 adj=63998632  heater_bed: target=115 temp=115.1 pwm=0.249 raspberry_pi: temp=49.7 mcu_temp: temp=38.6 CANBOARD: temp=75.0 chamber: temp=47.9 sysload=0.49 cputime=44231.390 memavail=414624 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=69.9 pwm=0.000
Traceback (most recent call last):
  File "/home/trident/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 303, in <module>

    main()
  File "/home/trident/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 289, in main
    axis = create_shaper_graph(keep_csv=options.keep_csv, max_smoothing=options.max_smoothing, scv=options.scv)
  File "/home/trident/klippain_shaketune/K-ShakeTune/scripts/is_workflow.py", line 123, in create_shaper_graph
    fig = shaper_calibration([new_file], KLIPPER_FOLDER, max_smoothing=max_smoothing, scv=scv)
  File "/home/trident/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 217, in shaper_calibration

    performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper(datas[0], max_smoothing, scv, max_freq)
  File "/home/trident/klippain_shaketune/K-ShakeTune/scripts/graph_shaper.py", line 54, in calibrate_shaper
    shaper, all_shapers = helper.find_best_shaper(
TypeError: find_best_shaper() got an unexpected keyword argument 'shapers'

[... cut ...] 6785295 bytes_read=6018239 bytes_retransmit=0 bytes_invalid=0 send_seq=335232 receive_seq=335232 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=2 upcoming_bytes=0 freq=63998559 adj=63998577  heater_bed: target=115 temp=115.0 pwm=0.293 raspberry_pi: temp=49.7 mcu_temp: temp=38.4 CANBOARD: temp=75.0 chamber: temp=47.9 sysload=0.53 cputime=44231.966 memavail=418772 print_time=5644.097 buffer_time=0.000 print_stall=0 extruder: target=0 temp=70.1 pwm=0.000
Command {shaketune} finished

tecno-roy, 2024-03-28:
The same problem on my machine...#82 
[https://github.com/Frix-x/klippain-shaketune/issues/82]

Frix-x, 2024-03-28:
This is the same problem indeed and is due to using a Shake&Tune v2.6+ with an older Klipper version. Please update Klipper to get rid of this or use S&T <=v2.5

einsteins12, 2024-03-28:
Thank You Felix

Once I get Klipper up to date and the MCU fixed for it, will I need to reinstall S&T ?

Thanks for your assistance!

 

Bill

Frix-x, 2024-03-28:
> Once I get Klipper up to date and the MCU fixed for it, will I need to reinstall S&T ?

No it should be ok, just update Klipper (and your MCUs) and everything should work again :)

closed 2024-04-11 by Frix-x