Frix-x/klippain-shaketune #84: AXES_MAP_CALIBRATION not outputing any information

TekuSP, 2024-04-03:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v2.6.1-1-gbf7a98d9

### Describe the bug and expected behavior

```
13:06:38 
//
13:06:38 
// Starting accelerometer axe_map calibration
13:06:38 
// This operation can not be interrupted by normal means. Hit the "emergency stop" button to stop it if needed
13:06:38 
//
13:06:38 
// accelerometer measurements started
13:06:45 
// Writing raw accelerometer data to /tmp/adxl345-axemap.csv file
13:06:45 
echo: Analysis of the movements...
13:06:45 
// Running Command {shaketune}...:
13:06:49 
// WARNING: AXES_MAP_CALIBRATION is currently very experimental and may produce incorrect results... Please validate the output!
13:06:50 
// Command {shaketune} finished
```

Command runs, but nothing is outputed other than warning.
Klipper version: v0.12.0-125-gbfb71bc2

xyleth, 2024-04-05:
I am seeing the same bug on a fresh install of shake and tune alongside Klippain.

If I dig in, activate the shake and tune PyEnv and then manually run the script I get output:

`./analyze_axesmap.py /tmp/adxl345-axemap.csv -a 1500
Detected axes_map:
  z,-y,x`

Beyond that my Klipper macro-fu isn't strong enough to pinpoint where this is getting lost, but hopefully it helps.

Frix-x, 2024-05-12:
Closing this in favor of #99 that is duplicate with more infos

closed 2024-05-12 by Frix-x