Frix-x/klippain-shaketune #95: Support hybrid_codexy machines

seacred, 2024-04-29:
### Describe the feature or hardware support you'd like

It would be great to support hybrid_corexy machines like the upcoming RatRig V-Core 4 or other builds like idex machines since they create much more vibrations due to the kinematics and the heavier x-axis.

Frix-x, 2024-04-30:
This is indeed planned. It should be pretty straightforward but I will need some testers as I don't own any machine of that kind yet

seacred, 2024-05-03:
hit me up if i can provide some testing.

Frix-x, 2024-05-05:
Can you tell me what is the angle that the toolhead is running at when using only X motors? Is it 135 degrees ?

seacred, 2024-05-10:
[image]

Do you mean if only one X motor ist moving?
A motor is just movin the T0 Toolhead left and right and B motor moves T1 left and right.

depending on the mode on IDEX for example B needs to invert A movement so the T1 can stay in place.
While copy mode both AB drives move the exact same.
If you move X axis on Y by hand towards infinite the toolheads move together. if you move the X axis towards Y0 they move apart.

DIYlover, 2024-10-16:
> This is indeed planned. It should be pretty straightforward but I will need some testers as I don't own any machine of that kind yet

I can help on testing too.  I do have a hybrid-coreXY IDEX printer even with AWD configuration for both A-motor and B-motor. :-p

mg197879, 2024-12-18:
I get the following error when trying to install:

biqu@BTT-CB1:~$ wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash
--2024-12-18 16:26:32--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
ERROR: The certificate of ‘raw.githubusercontent.com’ is not trusted.
ERROR: The certificate of ‘raw.githubusercontent.com’ doesn't have a known issuer.

Any suggestions? I have an IDEX hybrid_corexy machine with AWD on Y which has 4 steppers, AWD on both carriages which is another 4 steppers.

arsonists, 2025-08-27:
I agree. If you need any proof, I can provide it too.

Andi3D-tech, 2025-11-30:
Hello @Frix-x !

I am using this -> https://github.com/SnakeOilXY/klipper-extended-corexy-kinematic to get my hybrid corexy working.
I is a RatRig vCore3.2H running Klippain. config modified for hybrid (Stepper A, B, C, C1.

But i cant get ShakeTune to work.

CoreXY and CoreXZ kinematics required, extended_corexy found

Can you help me?

Grinsekatze447, 2026-09-26:
Hello Frix-x i jsut opened an issue, i have exaclty this setup and amusing the SnakeOilXY kinematiks. Printer is woprking doing first layer calibration, heatcalibration and well wanted to do shaketune.

to use it, do i really just need to change the following line to haver extended corexy in it to:
so it looks like this:?

if motors_config_parser.kinematics in {'corexy', 'limited_corexy', 'extended_corexy'}: