Frix-x/klippain-shaketune #96: Error running command after update to v3

KazachThi, 2024-04-29:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v3.0.0-1-g56a5502d

### Describe the bug and expected behavior

With both Compare Belts Responses and Create Vibrations Profile it starts the measurement and once it finishes measuring gets an error "Error running command {shaketune}"

No problems before updating to v3.

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain-shaketune/files/15153050/klippy.log)

Running Voron 2.4, Beacon(H) as accel chip on XOL.

AmshaSharam, 2024-04-29:
i think the paths inside the files are changed, chek inside de files, i have to change mines from ~/printer_data/config/K-ShakeTune/shaketune.sh) to command: ~/printer_data/config/scripts/K-ShakeTune/shaketune.sh

Frix-x, 2024-04-29:
Yes this is indeed normal. The path changed in S&T v3.0.0 and I will need
to push an update as well on this side. I'll do it quickly but in the
meantime, add to your overrides:

```
[gcode_shell_command shaketune]
command: ~/printer_data/config/scripts/K-ShakeTune/shaketune.sh
timeout: 600.0
verbose: True
```

KazachThi, 2024-04-29:
Thanks for the update. 

Will close !

closed 2024-04-29 by KazachThi