Frix-x/klippain-shaketune #97: Not showing *.png on interface

AmshaSharam, 2024-04-29:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

v3.0.0-1-g56a5502d

### Describe the bug and expected behavior

Can see the results image from belt test, i have to download then i can see on windows, any ideia why?

Frix-x, 2024-04-30:
Hello,

I know that Mainsail's image viewer can sometimes behave a bit strange and stretch/compress images.
FYI, Fluidd works fine (and that's what I use on my side).

I'll see with Mainsail team if something can be done.

Frix-x, 2024-04-30:
Oh, ok the issue I was thinking about was apparently solved at the beginning of the year: https://github.com/mainsail-crew/mainsail/pull/1746

Can you tell me more details or maybe add some screenshots about your issue?

tdlane1, 2024-05-11:
I was encountering this issue after a replacement of the SBC on my printer.  When trying to view the image in mainsail, only an image placeholder would be shown.  And error would occur if trying to download the image as well.  Of note:   Subfolders in K-ShakeTune_results were not appearing, only the various PNG images from the various tests (No separation of images into different folders)

I was using Brave browser, so I tested behavior in Edge and everything was working as expected.

Fix was to clear the browser cache.  No results images are showing in their respective subfolders and viewable.

closed 2024-05-12 by Frix-x