Frix-x/klippain-shaketune #99: AXES_MAP_CALIBRATION:  'AxesMapFinder' object has no attribute 'create_graph'

bart594, 2024-05-01:
### K-Shake&Tune module branch

- [X] I confirm using the main branch

### Version

Shake&Tune version: v3.0.0-1-g56a5502

### Describe the bug and expected behavior

If I run the macro from mainsail dashboard I'm getting errors, data are collected  but can't create graph

`Writing raw accelerometer data to /tmp/mpu9250-axemap.csv file
Running Command {shaketune}...:
MPU9250 finished 'mpu9250' measurements
Traceback (most recent call last):
  File "/home/pi/klippain_shaketune/src/is_workflow.py", line 408, in main
    graph_creator.create_graph()
AttributeError: 'AxesMapFinder' object has no attribute 'create_graph'

Shake&Tune version: v3.0.0-1-g56a5502
Warning: Failed to restore the original locale setting: unsupported locale setting
Error while generating the graphs: 'AxesMapFinder' object has no attribute 'create_graph'
Command {shaketune} finished
`

Data are collected in csv file but shaketune.sh script can't create graph. Maybe some python  libs are missing ? But i have ran install script once more and everything is ok (no errors or missing dependencies)

pi@mainsailos:~/printer_data/config/K-ShakeTune $ ./shaketune.sh  -t axesmap --chip_name mpu9025
Shake&Tune version: v3.0.0-1-g56a5502
Warning: Failed to restore the original locale setting: unsupported locale setting
Error while generating the graphs: 'AxesMapFinder' object has no attribute 'create_graph'
Traceback (most recent call last):
  File "/home/pi/klippain_shaketune/src/is_workflow.py", line 408, in main
    graph_creator.create_graph()
AttributeError: 'AxesMapFinder' object has no attribute 'create_graph'

Here csv file output

#time,accel_x,accel_y,accel_z
1140.624583,8216.285956,12768.552499,148201.174088
1140.625565,8142.265362,13101.645173,147894.735889
1140.626547,6920.925558,13101.645173,148622.526612
1140.627528,6254.740210,12287.418637,149426.926885
1140.628510,6217.729913,12176.387746,149350.317335
1140.629492,6143.709319,12472.470123,149618.450759
1140.630474,6106.699021,12139.377449,150078.108058
1140.631455,6365.771101,11880.305369,149656.755534
1140.632437,6217.729913,12324.428934,148814.050486
1140.633419,5810.616645,12879.583391,148584.221837
1140.634401,5884.637239,12879.583391,148316.088413
1140.635382,6143.709319,12768.552499,147779.821564

Pneumanifest, 2024-05-01:
Same issue for me. 
[image]

EightFinger, 2024-05-02:
Same for me too!

rallegade, 2024-05-03:
Same here after i updated.

Marucci87, 2024-05-09:
I have same issue, how do you get past this?

Marucci87, 2024-05-10:
I just ran create vibrations profile macro then got this afterwards (Shake&Tune version: v3.0.0-1-g56a5502
Error while generating the graphs: File /home/biqu/printer_data/config/K-ShakeTune_results/vibrations/vibrations_20240509_195220_vibrations-axemap.csv.csv does not match expected format. Clean your /tmp folder and start again!)

Bloodpack, 2024-05-12:
Same issue here, fresh install

mhaas, 2024-05-12:
Very stupid workaround:

```
voron@voron-02:~/klippain_shaketune$ git diff
diff --git a/src/is_workflow.py b/src/is_workflow.py
index 5ae1370..c037828 100755
--- a/src/is_workflow.py
+++ b/src/is_workflow.py
@@ -371,6 +371,9 @@ class AxesMapFinder:
         with result_filename.open('w') as f:
             f.write(results)
 
+    def create_graph(self):
+        return self.find_axesmap()
+
 
 def main():
     options = Config.parse_arguments()
```

mhaas, 2024-05-12:
FWIW, my patch above makes the actual error go away, but I still don't get any output because the CSV files in /tmp cannot be found - which doesn't make sense to me, because running the same code in Python shows that they are found. I'll stop playing with this now, but wanted to leave a note that the patch above is not enough.

Frix-x, 2024-05-12:
This is actually a known bug that was left as is for the v3.0.0 release, as this feature is not the most used and was already quite buggy... So I wanted to spend some time working on it to make it more reliable. In fact, it was almost like a random guess and definitely not perfect, so I think it's better to just not give anything than to give wrong results haha.

The macro call is already fixed for the next v4 release I'm working on. In the meantime, just set your axes_map manually, it should be no big deal. The rest should work as usual.

akadata, 2024-06-05:
did you merge this back into master branch? after recovering from installing your other repo and getting back to this working, which I may add does not work using your install script. I have this issue

1:09 AM
File "/home/pi/klippain_shaketune/src/is_workflow.py", line 408, in main
graph_creator.create_graph()
^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'AxesMapFinder' object has no attribute 'create_graph'
Shake&Tune version: v3.0.0-1-g56a5502
Error while generating the graphs: 'AxesMapFinder' object has no attribute 'create_graph'

P3T3R25, 2024-06-08:
> did you merge this back into master branch? after recovering from installing your other repo and getting back to this working, which I may add does not work using your install script. I have this [...]

I have the same issue and I do not know how to override it.
When I was doing it for the first time, you could override it by activating the script manually, but right now i does not work.
Does anyone know how to run it?

dezenten, 2024-06-08:
Same problem here, fresh install.

urbinek, 2024-06-09:
Same issue, fresh install 
```
$ AXES_MAP_CALIBRATION
18:28:03 //
18:28:03 // Starting accelerometer axe_map calibration
18:28:03 // This operation can not be interrupted by normal means. Hit the "emergency stop" button to stop it if needed
18:28:03 //
18:28:03 // accelerometer measurements started
18:28:10 // Writing raw accelerometer data to /tmp/adxl345-axemap.csv file
18:28:10 
echo: Analysis of the movements...
18:28:10 // Running Command {shaketune}...:
18:28:19 // Traceback (most recent call last):
// File "/home/biqu/klippain_shaketune/src/is_workflow.py", line 408, in main
// graph_creator.create_graph()
18:28:19 // AttributeError: 'AxesMapFinder' object has no attribute 'create_graph'
// Shake&Tune version: v3.0.0-1-g56a5502
// Error while generating the graphs: 'AxesMapFinder' object has no attribute 'create_graph'
18:28:20 // Command {shaketune} finished
```

Frix-x, 2024-06-09:
This is normal as Shake&Tune v4 is still not released (as stated in this comment: https://github.com/Frix-x/klippain-shaketune/issues/99#issuecomment-2106306394).

But a lot of work has been done on this issue and now the macro will be fixed and will have good and valid results + some more insights. Please be patient, I'm finalizing the last bits for the v4 release, but it should be out soon!

closed 2024-06-13 by Frix-x